ros-visualization/rviz #1350: rviz freeze with moveit, allow external comm enabled

ljulc, 2019-03-08:
not sure if it is rviz issue or moveit issue. But at the end, rviz freezes.

what I did.
roslaunch abb_robot_moveit_config demo.launch
check “Allow External Comm” in planning tab
move robot arm position with mouse interacting with interactive marker (try to move robot arm to unreachable position)
after some movement, rviz freezes, stop responding to anything
my setup
Ubuntu 18.04.2 LTS
ROS Melodic
rviz 1.13.1
everything else works fine. nothing special noticed in log file and bag file.

rhaschke, 2019-03-08:
Is this error reproducible on your side? Does it also occur, when “Allow External Comm”  is not checked?

rhaschke, 2019-03-12:
Looks like this is a bug in the MoveIt plugin.

rhaschke, 2019-03-13:
Hm. I wasn't able to reproduce this issue.

closed 2019-03-29 by rhaschke