ros-visualization/rviz #1351: Missing interactive markers

imcardle, 2019-03-13:
### Description
I'm running through the MoveIt! getting started tutorial but I'm not able to see the "interactive markers" in rViz.  I've clicked on the "Interact" button, as suggested in the tutorial - but still the markers don't show up.

### Environment
* OS Version: Ubuntu 18.04 (running in VirtualBox)
* ROS Distro: Melodic
* RViz: 1.13.1
* Qt: 5.9.5
* OGRE: 1.9.0

rhaschke, 2019-03-14:
Are you using the Panda robot of the tutorial or are you using your own robot model?

imcardle, 2019-03-14:
Hi @rhaschke, I'm using the Panda robot.  I'd like to use my own eventually - but I'm starting by sticking to the demo/tutorial until I'm confident everything is working correctly.

rhaschke, 2019-03-14:
I cannot reproduce the issue. On my side, MoveIt interactive markers work fine. To further narrow down the issue, could you please try to run the [(interactive) visualization tutorials](http://wiki.ros.org/visualization/Tutorials) and report whether you see those markers? Maybe it's an issue with your virtualbox configuration?

imcardle, 2019-03-14:
OK, I tried the interactive visualization tutorials that you mentioned.  The interactive markers are displayed correctly.  So, I guess this narrows it down to a problem with the MoveIt! tutorial, rather than the rViz tool?

rhaschke, 2019-03-14:
In this case, please file an issue on https://github.com/ros-planning/moveit.

closed 2019-03-14 by rhaschke