ros-visualization/rviz #1355: MovableText global translation only works in y-direction

Eruvae, 2019-03-17:
When using the MovableText class in rviz/ogre_helpers for a custom rviz plugin, translating the text with setGlobalTranslation only moved the text in y-direction. In line 577/578 of movable_text.cpp, the translation is used as follows:
```
Vector3 ppos = mParentNode->_getDerivedPosition() + Vector3::UNIT_Y * mGlobalTranslation;
```
By copying the class to my own project and removing Vector3::UNIT_Y from that line, the issue was fixed:
```
Vector3 ppos = mParentNode->_getDerivedPosition() + mGlobalTranslation;
```

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    rviz version 1.13.1
    compiled against Qt version 5.9.5
    compiled against OGRE version 1.9.0 (Ghadamon)
    OpenGl version: 3.1 (GLSL 1.4).
    ```

rhaschke, 2019-05-11:
Thanks for reporting this issue. Please consider next time to directly file a pull-request with the suggested change. This facilitates review and usually speeds up processing :wink:

closed 2019-05-11 by rhaschke

atasoglou, 2021-11-19:
Hey @Eruvae, thanks for catching this. I am having the same issue. I added it also to my project, compiles fine but I think RViZ is still loading the old one. Any ideas, pointers? Thanks!

Eruvae, 2021-11-19:
@atasoglou Since the issue was already fixed in #1375, this shouldn't be an issue anymore, unless you are using an old version of rviz.

atasoglou, 2021-11-19:
It seems we are :( Do you remember how you went around this at that point?

Eruvae, 2021-11-19:
I have the plugin I wrote with custom changes to the movable text here: https://github.com/Eruvae/hypermap_rviz_plugin
You can have a look. But basically, as long as your plugin includes your version of movable text instead of the rviz version, it should use it. I would recommend that you change the namespace, so that you can be sure it's the right one and there's no conflict if you include both.

atasoglou, 2021-11-19:
Thanks @Eruvae I will look into the project. I thought as well that rviz will pick up the fixed class. No clue I am gonna check again. And I also like the idea of the namespace! Thanks!

Eruvae, 2021-11-19:
Well, if you would want that change in rviz directly, you'd have to clone this repository and compile from source with the changes. However, that wouldn't really change anything, since rviz itself doesn't use the global translation of movable text.

atasoglou, 2021-11-19:
namespace change did the trick for me! :1st_place_medal: Thanks again!

PR #1375 (2019-05-11): MovableText: consider full mGlobalTranslation vector
MovableText: Consider full `mGlobalTranslation` vector and not only y-component. Fixes #1355.