ros-visualization/rviz #1359: Error: using typedef-name ‘Ogre::Vector3’ after ‘class’ class Vector3

senceryazici, 2019-03-27:
Hi I'm trying to build rviz on Mac OS X Mojave.
I'm using melodic as ros distro. I created the catkin workspace and built almost every bare bone melodic ros packages and some rqt packages inside the directory, and installed them on /opt/ros/melodic. 

I've ogre installed on my system and when building rviz the CMake finds the required packages like ogre and yaml etc..

But I get this error when building
```
error: definition of type 'Vector3' conflicts with typedef of the same name
class Vector3;
      ^
/usr/local/Cellar/ogre/1.7.4/include/OGRE/OgrePrerequisites.h:251:29: note: 'Vector3' declared here
    typedef Vector<3, Real> Vector3;
```
I think this is some kind of conflict with OGRE and rviz, but I don't know why this issue pop up on Mac only or even does it happen on Linux too, but if the definitions should be made different I can fork the repo and modify as you suggest me to do.

Also, I think errors pop up randomly due to too many errors, the compiler shows some of them, but sometimes i get the error of,

```
rviz/src/rviz/mesh_loader.cpp:62:10: fatal error: 'assimp/assimp.hpp' file not found
#include <assimp/assimp.hpp>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
```

I know rviz is not supported on Mac OS X and all the other ROS core packages are supported for Mac OS X experimentally. But it doesn't seem an unsolvable problem to me, of course I'm not a building expert as you would say. 
All in all, any kind of help is appreciated.

If the question has lack of information or unclear parts I can provide feedback and additional info as you request..

Thanks in advance,
Sincerely,
Sencer

### My system
 * Ros Distro: Melodic
 * OS: Mac OS X Mojave 10.14.3
 * Git Branch: release/melodic/rviz/1.13.3-0

rhaschke, 2019-03-27:
The Vector3 error is missing the error location.
I'm not a MacOS user, so I'm not sure I will be able to help...

senceryazici, 2019-03-27:
> The Vector3 error is missing the error location.
I don't think I understand this, can you explain it a little ?
Do you think the problem is spesific to Mac OS or else ?
I saw the same issue created time ago, but the issue was closed due to lack of feedback i think, anyway the poster was a linux user, so it is not related to Mac OS ?

rhaschke, 2019-03-28:
Please provide the full error message you've got.
From the stuff you posted, the error location is missing that the compiler was complaining about.

rhaschke, 2019-05-11:
Did you solve your problem? If so, please close this issue.
If not, please provide the requested information, so that we can try to help.
By the way, the melodic branch of rviz builds against Ogre 1.9.0, but you build against 1.7.4. Maybe that's an issue.

senceryazici, 2019-06-14:
I think I solved the Vector3 problem, as though I get different errors, 

but the assimp error, still exists

`/Users/senceryazici/catkin_ws/src/desktop_full_packages/desktop_full_extra/rviz/src/rviz/mesh_loader.cpp:62:10: fatal error: 'assimp/assimp.hpp' file not found
#include <assimp/assimp.hpp>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/rviz/CMakeFiles/rviz.dir/mesh_loader.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/rviz/CMakeFiles/rviz.dir/all] Error 2
make: *** [all] Error 2`

rhaschke, 2019-08-30:
Obviously, you need to install the assimp package.

closed 2019-08-30 by rhaschke

silyman1, 2019-10-24:
> By the way, the melodic branch of rviz builds against Ogre 1.9.0, but you build against 1.7.4. Maybe that's an issue.

hi,I want to know the melodic branch of rviz supoort which version of ogre

rhaschke, 2019-10-24:
@silyman1, melodic-devel supports Ogre 1.9.1 for sure.
noetic-devel supports recent Ogre releases up to 1.12.2

silyman1, 2019-10-24:
when I compile ogre 1.9.1 ,problems happened:
/home/toybrick/ogre-1-9-0/OgreMain/include/OgreStringConverter.h:126:23: error: 'static Ogre::String Ogre::StringConverter::toString(long unsigned int, short unsigned int, char, std::ios_base::fmtflags)' cannot be overloaded with 'static Ogre::String Ogre::StringConverter::toString(size_t, short unsigned int, char, std::ios_base::fmtflags)'
         static String toString(unsigned long val,
                       ^~~~~~~~
/home/toybrick/ogre-1-9-0/OgreMain/include/OgreStringConverter.h:122:23: note: previous declaration 'static Ogre::String Ogre::StringConverter::toString(size_t, short unsigned int, char, std::ios_base::fmtflags)'
         static String toString(size_t val,
                       ^~~~~~~~
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/build.make:89: OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:91: OgreMain/CMakeFiles/OgreMain.dir/all] Error 2

rhaschke, 2019-10-24:
Please don't report Ogre compile issues here. In any case, when I compiled different Ogre versions recently, I didn't have any specific issues.

npnmos, 2020-02-03:
Hello, sorry for posting in this old thread. 
@senceryazici How did you solve the problem about Vector3?
@silyman1 Even though this is not the place for discussing Ogre error, have you solved your problem?
Thank you all.

rhaschke, 2020-02-03:
This problem is solved on the noetic-devel branch, particularly with commit https://github.com/ros-visualization/rviz/commit/f9a671bd3f7b248e0543c52dd81e35796a70707c.

senceryazici, 2020-02-03:
Hi @npnmos 
I was compiling with a wrong version of OGRE. Make sure you are using the correct version

flynneva, 2020-06-13:
im still seeing this error while building for noetic with ogre 1.12:

```
In file included from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/frame_manager.h:41,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/display_context.h:37,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/message_filter_display.h:37,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/../../../../../src/visualization_tutorials/rviz_plugin_tutorials/src/imu_display.h:36,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/moc_imu_display.cpp:9,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/mocs_compilation.cpp:3:
/usr/include/OGRE/OgreVector3.h:2:62: note: #pragma message: /usr/include/OGRE/OgreVector3.h is deprecated, migrate to Ogre.h
    2 | #pragma message( __FILE__ " is deprecated, migrate to Ogre.h")
      |                                                              ^
In file included from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/moc_plant_flag_tool.cpp:9,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/mocs_compilation.cpp:4:
/home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/../../../../../src/visualization_tutorials/rviz_plugin_tutorials/src/plant_flag_tool.h:37:7: error: using typedef-name ‘Ogre::Vector3’ after ‘class’
   37 | class Vector3;
      |       ^~~~~~~
In file included from /usr/include/OGRE/OgreVector.h:32,
                 from /usr/include/OGRE/OgreVector3.h:1,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/frame_manager.h:41,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/display_context.h:37,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/message_filter_display.h:37,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/../../../../../src/visualization_tutorials/rviz_plugin_tutorials/src/imu_display.h:36,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/moc_imu_display.cpp:9,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/mocs_compilation.cpp:3:
/usr/include/OGRE/OgrePrerequisites.h:253:29: note: ‘Ogre::Vector3’ has a previous declaration here
  253 |     typedef Vector<3, Real> Vector3;
      |                             ^~~~~~~
```