ros-visualization/rviz #1364: Support ascii STL files

simonschmeisser, 2019-04-05:
rviz currently uses handcrafted code for loading STLs and assimp for everything else
https://github.com/ros-visualization/rviz/blob/74334b765e0eea6d1de371ae8a9c32d769e51e4d/src/rviz/mesh_loader.cpp#L700

does anybody know why? I'm asking because this handcrafted code does not support ASCII format of STL files which creates a lot of confusion among our test users/customers. I will open a PR to simply use assimp for everything

rhaschke, 2019-04-05:
@wjwwood, @jfaust, do you remember the reason for not using assimp for STLs?
Looking into the commit history, it looks like STL loading was there first and later assimp was added to support more formats as well. So, if assimp can handle all formats, this proposal looks reasonable to me.

wjwwood, 2019-04-05:
I don't know, if code can be deleted for assimp instead, that's fine with me. Though something like this should be tested well (maybe even ask the community to test is specifically).

jfaust, 2019-04-05:
The commit in question: https://github.com/ros-visualization/rviz/commit/3117a98853b5afe8d3d1ba54308a8e70b58033e6

That's from 9 years ago, so I don't remember specifically, but based on my experience with assimp in general, I would guess it choked on some of the STL files we had at the time, and it was easier to write our own parser than to fix theirs.

rhaschke, 2019-08-31:
Addressed in #1365.

closed 2019-08-31 by rhaschke