ros-visualization/rviz #1368: Docking camera displays causes bad "re-enable" behavior, and can "ruin" QMainWindow state

EricCousineau-TRI, 2019-04-23:
I am tinkering with RViz on ROS1 (with an eye towards using ROS2 at some point in the future).
I've found that the tabbed display features of RViz to be a bit buggy.

Possibly relates #940

To reproduce, if you have 2+ Camera displays, and you try to stack them, then RViz struggles to show one of the windows.

Workflow, using this code:
https://github.com/EricCousineau-TRI/repro/tree/c67563df222c2aa035965888bb832f7c7267990f/ros/ros1_camera_widget_bug

1. I start with this:
> [image]
2. I drag `/camera/1` on top of `/camera/0`, which hides `/camera/1`; `/camera/1` is now disabled:
> [image]
3. I try to re-enable `/camera/1`; it only flickers, but it stays hidden.
4. I hide `/camera/0`, and then re-enable `/camera/1`; eh, it shows that:
> [image]
5. Now I click `/camera/0` to show it again, and the tab shows (finally!). However, it's still disabled:
> [image]
6. I click `/camera/0`s Enable checkbox, and it enables:
> [image]
7. I now click on its tab, and it now disables and hides `/camera/1`:
> [image]

I have also run into issues where it prints the following, then hangs (even after sending SIGINT / SIGABRT through the keyboard) + segfaults:
```
QPainter::begin: Paint device returned engine == 0, type: 2                                                                                               
QPainter::initFrom: Painter not active, aborted                                                                                                           
QPainter::save: Painter not active                                                                                                                        
QPainter::setPen: Painter not active                                                                                                                      
QPainter::setPen: Painter not active                                                                                                                      
QPainter::setRenderHint: Painter must be active to set rendering hints                                                                                    
QPainter::translate: Painter not active                                                                                                                   
QPainter::setPen: Painter not active                                                                                                                      
QPainter::save: Painter not active                                                                                                                        
QPainter::setClipRect: Painter not active                                                                                                                 
QPainter::setBrush: Painter not active                                                                                                                    
QPainter::setBrush: Painter not active                                                                                                                    
QPainter::setPen: Painter not active                                                                                                                      
QPainter::restore: Unbalanced save/restore                                                                                                                
QPainter::restore: Unbalanced save/restore                                                                                                                
QPainter::end: Painter not active, aborted
^C^C^C^C^\^\Segmentation fault (core dumped)
```

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1556044860.940732975]: rviz version 1.13.1
    [ INFO] [1556044860.940829161]: compiled against Qt version 5.9.5
    [ INFO] [1556044860.940870230]: compiled against OGRE version 1.9.0 (Ghadamon)
    ```

EricCousineau-TRI, 2019-04-23:
The workaround in this case is to either:
a. Disable one of the cameras and drag it elsewhere; or
b. Restore or delete `QMainWindow State` from `config.rviz` - but blech!

rhaschke, 2019-08-30:
@EricCousineau-TRI, sorry for the late response and many thanks for the detailed issue description!
This was very helpful. I think, I fixed the issue: https://github.com/ros-visualization/rviz/pull/1409.
Could you give it a try?

EricCousineau-TRI, 2019-08-30:
Yup, that seemed to work!

Repro steps:
```sh
$ git clone https://github.com/ros-visualization/rviz.git && cd rviz
$ git fetch origin pull/1409/head & git checkout FETCH_HEAD
HEAD is now at cafa965d debug Travis
$ source /opt/ros/melodic/setup.bash
$ mkdir build && cd build && cmake ..
$ make -j
$ source ./devel/setup.bash
$ ./devel/lib/rviz/rviz -d {...}/repro/ros/ros1_camera_widget_bug/config.rviz
```

Now when I drag `/camera/1` on top of `/camera/0`, 1 is shown and 0 is hidden. Enabling / disabling / switching through tabs does not produce any unexpected results.
Thanks!

PR #1409 (2019-08-30): Fix several issues with Camera and Image display
This PR fixes several, partially long-standing, issues with Camera and Image display. See individual commit messages for details.

Fixes #940 
Fixes #1368 

Unfortunately, the current, "clean" implementation requires an ABI change.
But I added an ABI compatible solution, using a global map to store the new member instead.
closed 2019-08-30 by rhaschke