ros-visualization/rviz #1369: Qt5 dependency on Kinetic - Cannot build with Qt4

theobslhc, 2019-04-24:
rviz will not build with Qt4 on branch **kinetic-devel** : 
```
.../rviz/src/rviz/visualization_frame.cpp:1253:66: error: ‘FindDirectChildrenOnly’ is not a member of ‘Qt’
   QList<QTabBar *> tab_bars = findChildren<QTabBar *>(QString(), Qt::FindDirectChildrenOnly);
```
[Merged PR #1168](https://github.com/ros-visualization/rviz/pull/1168) relies on `Qt::FindDirectChildrenOnly` to get a list of all `QTabBar` children of `VisualizationFrame`, which is a flag to `QObject::findChild()` that was introduced in Qt5.

See `visualization_frame.cpp`, line 1253 :  
```cpp
QList<QTabBar *> tab_bars = findChildren<QTabBar *>(QString(), Qt::FindDirectChildrenOnly);
```
Qt4 does not seem to provide an "easy" equivalent of this flag. The subject of reproducing its behaviour has been covered [here](https://stackoverflow.com/questions/8740278/is-there-a-simple-way-of-limiting-qobjectfindchild-to-direct-children-only), but proposed solutions rely on the children's object names, which I am not sure will be reliable in the context of rviz.

### Your environment
* OS Version: e.g. Ubuntu 16.04
* ROS Distro: Kinetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz: 
    ```
    no build, Qt 4.8.7
    ```

wjwwood, 2019-04-24:
Technically Kinetic requires Qt5:

http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021

Though we had tried to be friendly to Qt4 in that release, it's technically unsupported. So I can't say I'm surprised that it's broken.

theobslhc, 2019-04-24:
@wjwwood thanks for your feedback. I had missed that detail.
I will try to investigate the issue myself and submit a PR if time allows, for the sake of Qt4-friendliness :)

PR #1371 (2019-04-25): Fix compilation with Qt4 (#1369)
Fixes #1369.

This fix does not revert dock text eliding disabling as introduced in #1168 for Qt4.
Tested with Qt4 v4.8.7.
closed 2019-04-25 by theobslhc