ros-visualization/rviz #1372: Camera display type crashes rviz (Melodic) when deselected or deleted

ashnap123, 2019-04-29:
Camera display type crashes rviz when it it deselected or deleted. Recreated with multiple cameras on multiple machines.

### To recreate 
- Add any camera with a valid calibration
[image]

- Camera display image should be displayed with overlay
[image]

- Either delete or de-select Camera display to cause crash.
[image]

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    rviz version 1.13.3
    compiled against Qt version 5.9.5
    compiled against OGRE version 1.9.0 (Ghadamon)
    Stereo is NOT SUPPORTED
    OpenGl version: 4.6 (GLSL 4.6).
    
* Console output
```
Gtk-Message: 12:27:56.341: GtkDialog mapped without a transient parent. This is discouraged.
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[1]    25031 abort (core dumped)  rviz
```

* Install type 
    Aptitude

mcsheehan, 2019-06-25:
I'd like to bump this as a fairly major issue. The program crashes on moving camera windows around.

rhaschke, 2019-08-13:
Upstream issue: https://github.com/ros/geometry2/pull/402.

closed 2019-08-27 by rhaschke

tsukione, 2021-09-24:
after 25d2c81cfd6a8660a7dfb953cfd67e5659cbeede this issue reappear @rhaschke

PR #1404 (2019-08-27): Fix segfault in Camera plugin
Fixes #1372. As https://github.com/ros/geometry2/pull/402 isn't released yet, I'm going to merge this quick fix here. Thanks to @hexonxons, who analyzed the bug in detail in #1393 and originally proposed a solution.