ros-visualization/rviz #1377: fatal error: qregularexpression.h: No such file or directory

dclanmaster, 2019-05-26:
Describe your issue here and explain how to reproduce it.
I compiled the code with qtcreator, cmake has been through, but when run debug, this error always comes. But before, there has never been such issue.

### Your environment
* OS Version: Ubuntu 16.04
* ROS Distro: [Kinetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```
19:30:49: Running steps for project rviz...
19:30:49: Starting: "/usr/bin/make" 
[  0%] Automatic moc for target executable
[  0%] Built target executable_automoc
[  0%] Automatic moc for target rviz
[  0%] Built target rviz_automoc
[ 47%] Built target rviz
[ 48%] Built target executable
[ 48%] Automatic moc for target rviz_default_plugin
[ 48%] Built target rviz_default_plugin_automoc
[ 80%] Built target rviz_default_plugin
[ 80%] Automatic moc for target rviz_image_view
[ 80%] Built target rviz_image_view_automoc
[ 83%] Built target rviz_image_view
[ 83%] Automatic moc for target render_points_test
[ 83%] Built target render_points_test_automoc
[ 86%] Built target render_points_test
[ 86%] Automatic moc for target line_edit_with_button_test
[ 86%] Built target line_edit_with_button_test_automoc
[ 87%] Built target line_edit_with_button_test
[ 88%] Automatic moc for target connect_test
[ 88%] Built target connect_test_automoc
[ 90%] Built target connect_test
[ 90%] Automatic moc for target new_display_dialog_test
[ 90%] Built target new_display_dialog_test_automoc
[ 91%] Built target new_display_dialog_test
[ 91%] Automatic moc for target render_panel_test
[ 91%] Built target render_panel_test_automoc
[ 93%] Built target render_panel_test
[ 93%] Automatic moc for target interactive_marker_test
[ 93%] Built target interactive_marker_test_automoc
[ 94%] Built target interactive_marker_test
[ 94%] Automatic moc for target color_editor_test
[ 94%] Built target color_editor_test_automoc
[ 95%] Built target color_editor_test
[ 97%] Automatic moc for target two_render_widgets
[ 97%] Built target two_render_widgets_automoc
[ 98%] Built target two_render_widgets
[100%] Compiling generated code for rviz_sip Python bindings...
/usr/share/sip/PyQt5/QtCore/qregularexpression.sip:26:32: fatal error: qregularexpression.h: No such file or directory
compilation terminated.
Makefile:18: recipe for target 'siplibrviz_siprvizToolManager.o' failed
src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:60: recipe for target 'devel/lib/python2.7/dist-packages/rviz/librviz_sip.so' failed
CMakeFiles/Makefile2:5234: recipe for target 'src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all' failed
Makefile:138: recipe for target 'all' failed
make[3]: *** [siplibrviz_siprvizToolManager.o] Error 1
make[2]: *** [devel/lib/python2.7/dist-packages/rviz/librviz_sip.so] Error 2
make[1]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
make: *** [all] Error 2
19:30:50: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project rviz (kit: Desktop)
When executing step "Make"
19:30:50: Elapsed time: 00:01.
```
* kinetic-devel

rhaschke, 2019-05-26:
Are you sure, you have all (Qt) dependencies correctly installed?

dclanmaster, 2019-05-27:
> Are you sure, you have all (Qt) dependencies correctly installed?

Dear rhaschke,
Thank you for your reply.
I'm really not sure, if Qt dependencies are correctly installed.
The procedure is simply:
install of ROS, and then,
```
sudo apt-get install qtcreator.
```
If there is something wrong, please tell me which Qt dependencies are additionally needed.

rhaschke, 2019-05-27:
This is not sufficient at all. qtcreator is just an IDE. In your catkin workspace, you should install required dependencies like so:
`rosdep install --from-paths . --ignore-src --rosdistro $ROS_DISTRO`

dclanmaster, 2019-05-27:
Sorry for not mentioning that, I did “rosdep install” before catkin make. 
This error only shows when I try to run debugger in Qtvreator. 
Cmake has been through actually.

rhaschke, 2019-05-27:
You are still mixing up things: Your reported error is a _compiler_ error. It's not related to a debugger.
Can you successfully compile on the cmdline? If so, probably your project configuration in Qt is wrong.
There are several tutorials on the web, how to use qtcreator for ROS projects. Essentially, it boils down to building with catkin on the cmdline _first_, and only afterwards loading the project into qtcreator as a (preconfigured) cmake project, referring to the catkin-created build path.
I'm closing this here, as it is definitely not an rviz issue. Please pose your questions on ROS answers.

closed 2019-05-27 by rhaschke