ros-visualization/rviz #1379: I am using Velodyne VLP16. I have configured the LIDAR to 1000rpm. I was able to get the data through the ros-kinetic-velodyne package till rescently. It used to publish the pointcloud on the /velodyne_points topic after the connection is established and with this lauch file "roslaunch velodyne_pointcloud VLP16_points.launch"

prabinrath, 2019-06-11:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Kinetic | Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    paste your console output here
    ```
* If source build, which git commit?

prabinrath, 2019-06-11:
The connection is intact and I am able to see the web GUI through the browser. The data used to com properly but its weird why its not coming any more. The /velodyne_packets topic is also blank. Echoing the /velodyne_packets and /velodyne_points prints nothing. I was able top get the data with this exact procedure earlier. I can still access the web gui but not the data through the ros topic

atyshka, 2019-07-23:
This is clearly not a problem with RViz, please move your issue to something like ROS anwers or the vlp-16 repository issues.

closed 2019-08-29 by rhaschke