ros-visualization/rviz #1380: Show further properties for selected Markers

B0gdar, 2019-06-16:
## 1. Brief description
When a Marker is being selected with the selection tools just [shows position and orientation values](https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/default_plugin/markers/marker_selection_handler.cpp#L71-L83).

* **Cube**
[image]
* **Sphere**
[image]
* **Cylinder**
[image]

## 2. Explanation

When a Marker is being selected with the selection tools just [shows position and orientation values](https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/default_plugin/markers/marker_selection_handler.cpp#L71-L83).

In general is enough, but when inspecting volumes or shapes it would be nicer to being able to straightforward inspect the scale as well (p.e. Bounding box fitting applications). Current state just make a menu with entries for getPosition() and getOrientation(). This could be expanded based on marker type and show additional info.

Changes needed to introduce this feature are small, I am not so sure about possible impact with interactive markers if any. I would try to make the changes myself when I find some spot, but everybody is welcome to perform them.

**I would also like to encourage other ROS users that do not know the selection tool to actively make use of it, as it saves time with respect the usage of other ROS packages or external tools.**

 The selection tool is simple but I strongly believe that if a broader audience will make use of the tool, cooler features would be able to be introduced 😄

## 3. Which distros are affected
All distros from indigo, as this tool was introduced long time ago.

## 4. To reproduce the issue

Reproduction is simple, just publish a marker with rostopic (CUBE, type:1; SPHERE, type:2; CYLINDER, type:3):
```bash
rostopic pub /shape visualization_msgs/Marker "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: 'map'
ns: 'MarkerWithShape'
id: 0
type: 1
action: 0
pose:
  position: {x: 0.0, y: 0.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}
scale: {x: 20.0, y: 10.0, z: 5.0}
color: {r: 1.0, g: 0.0, b: 0.0, a: 1.0}
lifetime: {secs: 0, nsecs: 0}
frame_locked: false
points:
- {x: 0.0, y: 0.0, z: 0.0}
colors:
- {r: 1.0, g: 0.0, b: 0.0, a: 1.0}
text: ''
mesh_resource: ''
mesh_use_embedded_materials: false" 
```
Open in rviz, enable selection panel, add marker to /shape topic and select the marker.

B0gdar, 2019-06-16:
Propossed tag: Enhacement

rhaschke, 2019-08-29:
You are welcome to file a pull request :wink:

rhaschke, 2019-11-14:
Addressed in #1436.

closed 2019-11-14 by rhaschke