ros-visualization/rviz #1381: RViz crashes on start on OSX 10.14.5

samlcharreyron, 2019-06-24:
I've compiled ROS melodic from source on Mac OSX but I can't get RViz running. The program crashes on start (before even printing any info to the console). I've attached a stack trace from the crash report.

### Environment
* OS Version: Mac OSX 10.14.5
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
rviz: 1.12.16
Qt: 5.12.3
OGRE: 1.9
opengl: 1.0.0

[rviz_crash_report.txt](https://github.com/ros-visualization/rviz/files/3321037/rviz_crash_report.txt)

rhaschke, 2019-08-29:
I'm not an OSX user, but looking into your crash report, it looks like Qt already crashes when initializing the `QApplication`. Are you able to run other self-build Qt applications?
Why aren't you building rviz 1.13.x for Melodic?

samlcharreyron, 2019-08-30:
I am able to compile and run the following Qt example application
https://code.qt.io/cgit/qt/qtbase.git/tree/examples/widgets/mainwindows/application?h=5.13

I think I first tried the default melodic build with 1.13 and then downgraded to lunar. Both have the same issue.

rhaschke, 2019-08-30:
No idea then. Sorry.

Tobias-Fischer, 2020-12-06:
Hi @samlcharreyron,
Please check out https://github.com/RoboStack/ros-noetic (if you have questions, drop by at https://gitter.im/RoboStack/). We provide rviz for osx based on coda-forge.