ros-visualization/rviz #1382: Problems with the QT_INSTALL_LIBS environment var during build.

bionade24, 2019-06-25:
Hi, I'm the AUR ros melodic maintainer and many people are facing during build the issue with the latest rviz release (1.13.3), that the generated Makefile uses `QT_INSTALL_LIBS`, but this var is not set on Arch.
```
g++: error: 0/libQt5Core.so: No such file or directory
make[3]: *** [Makefile:38: "/home/user/.cache/yay/ros-melodic-rviz/src/build/devel/lib/python3.7/site-packages/rviz/librviz_sip".so] Error 1
make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:61: devel/lib/python3.7/site-packages/rviz/librviz_sip.so] Error 2
make[1]: *** [CMakeFiles/Makefile2:4935: src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
When you set `QT_INSTALL_LIBS` to `"/usr/lib"` or `'/usr/lib'`, `/bin/sh` returns 1, because it can't handle the patterns.
```
/bin/sh: /usr/lib: syntax error: operand expected (error token is "/usr/lib")
```
Is the second issue from a shell script without shebang or from the Makefile itself?

calvertdw, 2019-06-30:
I fixed it by doing the following on Arch:
- Run `colcon build --symlink-install` on ros2.repos master
- Get the error above
- Edit `build/qt_gui_cpp/sip/qt_gui_cpp_sip/Makefile:12` replacing `$$[QT_INSTALL_LIBS]` with `/usr/lib`
- Running `colcon` build again

bionade24, 2019-06-30:
@calvertdw [We already did this](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ros-melodic-rviz), but I don't want to have such a messy hack in my PKGBUILDs. This has to be fixed upstream but I don't have the time currently to find out what is causing this. Will look after it next week.

jwhendy, 2019-06-30:
Is [this](https://github.com/ros-visualization/python_qt_binding/pull/64) related?

bionade24, 2019-07-01:
Seems so. Hope it get's merged soon. Or should we patch it manually?

jwhendy, 2019-07-01:
I'd defer to someone more experienced to interpret that PR. To me it looks equivalent to us using `sed`, just done a bit higher up in the process. My worry is that we're all working around some higher root cause (which I now see you alluding to above as well).

Perusing the link from that linked issue sounds like sip5 is coming and will change all of this anyway. Maybe we keep the `sed` fix and see what the rest of the ROS team does to adapt to this change. I'm sure more will run into it!

bionade24, 2019-07-01:
[This](http://python.6.x6.nabble.com/Incorrect-Makefile-using-prl-files-from-Qt-5-12-4-td5255962.html) seems to be the root cause for our problem.

jwhendy, 2019-07-01:
I should have direct linked there. That's what I meant via:
>Perusing the link from that linked issue

Also, this is why I thought maybe we should just wait for sip5, as I think a bunch of ROS will have to adapt, and we can observe what the best way forward is.

calvertdw, 2019-08-06:
https://github.com/ros-visualization/python_qt_binding/pull/64 has been merged.

rhaschke, 2019-08-30:
Should have been fixed via https://github.com/ros-visualization/python_qt_binding/pull/64.

closed 2019-08-30 by rhaschke