ros-visualization/rviz #1384: Problem with rosrun rviz under ROSOnWindows

ryo-matsuzaka, 2019-07-10:
Hello.This is the first time to ask a question toward Github community.
I recently started to use ROSOnWindows.

In order to launch Rvitz, I typed 'rosrun rviz rviz'. Then, I could open rviz. But, I got the bellow error in the command prompt. Also, several windows appears.

Could you tell me how to solve the problem?

////////////////////////////////////////////////////////////////////////////////////////////////////
[ERROR] [1562757974.232382400]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Failed to load library C:/opt/ros/melodic/x64\bin/rviz_default_plugin.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/opt/ros/melodic/x64\bin/rviz_default_plugin.dll)
・
・
・
////////////////////////////////////////////////////////////////////////////////////////////////////

[image]

ryo-matsuzaka, 2019-07-11:
I could solved the problem by installing libraries listed in https://github.com/ms-iot/ROSOnWindows/issues/44

closed 2019-07-11 by rhaschke

Smitty-44, 2020-05-16:
I am getting this exact same error and messages.  I already have QT_PLUGIN_PATH defined.  It doesn't seem to have anything to do with QT.  Also, after hitting <OK> for two of those messages shown above (POCO exception, etc.), when rviz opens, it says, default.rviz -RViz, in the upper left corner of the display.
I checked in plugin_description.xml and the first line states: <library path="librviz_default_plugin">
I don't know if that is a problem or not.
Also, the list of 16 error messages corresponds to the 16 entries in file default.rviz.
I'm kind of stuck here.  I'm not sure what else to match up or check.

I have RosOnWindows as well.  And I have been using ROS for about 6 months now.
I have found that this similar issue has happened over the last 4 years in linux as well.
But some of that had to do with the change to 64 bit.
http://barelywalking.com/?p=246

I commented on this issue here because maybe it shouldn't be closed if its not a QT problem.

Any help would be appreciated. 
Thanks, Smitty