ros-visualization/rviz #1385: For collada meshes alpha of multiple RobotModel interfere

simonschmeisser, 2019-07-12:
We noticed today that adding a second RobotModel display (or derived MoveIt ones) with alpha = 0.2 changes the display of the primary RobotModel to also have transparency. This only happens for Collada meshes, not STL

I first enable one RobotModel with alpha = 1, the robot uses individual collada meshes while gripper and camera are stl meshes

[image]

now I enabled a second RobotModel where alpha=0.2 and it will (visually) override the value for both (the input field is not affected)

[image]

Notice how both robots now have alpha=0.2 except for the gripper and camera

I suspect this is due to non unique material names, any idea @rhaschke ?

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1562941360.215562081]: rviz version 1.13.3
[ INFO] [1562941360.215614003]: compiled against Qt version 5.9.5
[ INFO] [1562941360.215624573]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1562941360.431414141]: Stereo is NOT SUPPORTED
[ INFO] [1562941360.432597198]: OpenGl version: 3 (GLSL 1.3).
    ```

rhaschke, 2019-07-12:
Could you please check whether the issue was introduced in #1294, i.e. validate the version before this merge?

simonschmeisser, 2019-07-12:
[image]

seems to have regressed after commit c7b2daa13f0a6bcd2b7824329c0f40857090d0bf  

I disabled some links from both RobotModels to show that two alpha values are supported

PR #1387 (2019-07-16): Fix opacity of Collada meshes
Fixes an issue introduced in #1294 (e906f82eb2ef04e63dea03f834ae722f144bc712):
Materials of Collada mesh models need to be cloned for each instance of a robot link.
Fixes #1385. @simonschmeisser, could you please verify?
closed 2019-07-16 by rhaschke