ros-visualization/rviz #1386: Memory leak in kinetic

aPonza, 2019-07-15:
I'm launching `roslaunch own_pkg calibration.launch` which includes:
- easy_handeye's calibrate.launch;
- realsense2_camera's rs_rgbd.launch;
- aruco_ros's single.launch;
- panda_moveit_config's panda_control_moveit_rviz;
- other minor things.

What I see in `htop` is the RAM filling up (in the arc of at most ~5m, even though the next picture says 22m), eventually using swap and finally crashing the pc. I also tried with 8GB more RAM just to see if it was a stabilization sort of issue but the behaviour is the same. Ordering the processes by memory used I see the problem is happening in something related to rviz:

[image]

The behaviour seems like it's a memory leak, and looking for memory leak PRs in RViz I saw #1294 (among other things which are already included in my version) and wanting to test if that fixes the leak I cloned the repo, checked out melodic-devel and `git rebase -i kinetic-devel` to get a compilable version for kinetic with the things added by the referenced PR (I couldn't manage to cherry-pick...).
```
pick 29ecde8 enable Travis again
pick 1bef2fb fix various catkin_lint issues
pick 26e920e remove executable flag from .ttf files
pick 0e8c842 suppress spurious warnings
pick a9d63cb update maintainer list
pick 707dc6f suppress cmake warnings
pick f230eab remove support for tf < 1.11.3
pick 6a769aa XML parse error message added (#1317)
pick 47af93a Revert "support multiple material for one link (#1079)"
pick e906f82 simplify material handling
pick 28fb778 support different materials for individual visuals
pick 83ac836 work-around for buggy urdf parser for visual materials
```
After fixing a compilation error (something with tinyxml2's Error() method, I used the compiler's suggestion) I re-launched my launch file and it seems to still be a problem, so unless I messed up something, this is not the issue I have.

I can provide the launch file and anything else needed, in case, but I'd rather have some help with what I can test to figure out where the root cause of the problem lies. 

Also of note, last week I was seeing a `QXcbWindow: Unhandled client message: "_COMPIZ_TOOLKIT_ACTION"` in the log, which I'm not seeing anymore though.

### Your environment
* OS Version: Ubuntu 16.04
* ROS Distro: Kinetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
current version from apt which is 1.12.17-0xenial-20190608-123836-0800, with OpenGl version: 3 (GLSL 1.3)

rhaschke, 2019-07-15:
If you suspect rviz, verify the following:
- Without rviz running, you shouldn't see the memory leakage anymore, right?
- Then, start vanilla rviz and add one display after the other.
If you have found out, which display is the culprit, report back ;-)

aPonza, 2019-07-16:
I can confirm no issues if RViz is not started. Also while doing this I found the leak isn't triggered if the master is born out of roslaunch instead of independently started via `roscore`. The culprit seems to be MotionPlanning (there's only TF other than that in the rviz.config that easy_handeye is using).

Also these are displayed when adding that display:
```

[ INFO] - [/rviz_1563259975465041482::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
[ INFO] - [/rviz_1563259975465041482::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
[ INFO] - [/rviz_1563259975465041482::PlanningSceneMonitor::startSceneMonitor::949]: Starting scene monitor
[ INFO] - [/rviz_1563259975465041482::PlanningSceneMonitor::startSceneMonitor::955]: Listening to '/move_group/monitored_planning_scene'
[ INFO] - [/rviz_1563259975465041482::RobotInteraction::decideActiveComponents::97]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] - [/rviz_1563259975465041482::RobotInteraction::decideActiveComponents::97]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
```
I just checked about the hand config, and it's tied to using `panda_arm_hand` instead of `panda_arm` in panda_moveit config (we were talking about it [here](https://github.com/ros-planning/panda_moveit_config/issues/17), so it seems it does have an impact, and I had not fixed it yet). No clue why it seems to be loading the move group twice, is it expected?

rhaschke, 2019-07-16:
I'm afraid you need to narrow down the issue even more. When running panda_moveit_config's demo.launch, I don't see a continuous increase in memory, even over minutes. 
What I see, though, is an increase of 2MBytes in the first few minutes (probably due to TF buffers filling up). Then the memory usage settles.

aPonza, 2019-07-17:
I too don't see a leak with demo.launch, I left it on overnight and in the now it is using the same memory as it was at the start. I'll do more tests on my side and let you know.

aPonza, 2019-07-25:
The issue isn't there anymore... The only relevant changes are in easy_handeye, among them I moved a dummy publisher from using tf to using tf2. Not sure what else could be the root cause.

closed 2019-07-25 by aPonza