ros-visualization/rviz #1389: did somebody get working RVIZ in RPi4? (Raspbian Buster)

FPSychotic, 2019-07-19:
before enter in more detail, to try filter this a little, somebody has rviz working with RPi4. 
Just make note there is a issue with libboost1.67 and libogre 1.19. Finally I got install it but is giving a error with q5ct and segmetation fault.
if there is no positive answer I will detail this better as I cannot right now as I need reinstall the SO.
thank you!

rhaschke, 2019-08-29:
Currently, we are using Ogre 1.9.0. There is an [open PR](https://github.com/ros-visualization/rviz/pull/1242) of people working on Ogre 1.11 compatibility.

FPSychotic, 2019-08-29:
https://www.facebook.com/groups/353575658687628/permalink/369364973775363/
i made it work teo months ago

closed 2019-08-29 by rhaschke