ros-visualization/rviz #1394: rviz crashes when displaying faulty marker message

mvieth, 2019-08-01:
When displaying a marker, rviz crashes with the message:
`rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp:630: virtual void Ogre::Node::setScale(const Ogre::Vector3&): Assertion !inScale.isNaN() && "Invalid vector supplied as parameter"' failed.`
One of the values in the field `scale` in the marker has a very extreme value (6.34119308747e+180). This is of course itself an error, but rviz should catch that error instead of crashing. I think what happens is that the `validateFloats` in `marker_display.cpp` does not complain because it is a vector of doubles, but the `Real`-type in Ogre is a float, which cannot hold this large value, so the assertion in `setScale` fails. I am not sure how to best fix this issue, so I cannot provide a pull request.

To reproduce: display a marker topic in rviz and publish a faulty marker message with rostopic

Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 16.04
* ROS Distro: Kinetic (but probably also a problem in Melodic)
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    rviz version 1.12.17
    compiled against Qt version 5.5.1
    compiled against OGRE version 1.9.0 (Ghadamon)
    Stereo is NOT SUPPORTED
    OpenGl version: 4.6 (GLSL 4.6).
    ```
* If source build, which git commit? latest kinetic-devel

christian-rauch, 2019-08-06:
I also encounter RViz crashes related to the same assertion in `Ogre::Node::setScale` on melodic. But I do not see any suspicious large values in my markers. I agree that these issues should be catches by the plugin that processes the values and present this more nicely to the user without crashing all of RViz.

mvieth, 2019-08-06:
@christian-rauch interesting. Could you post the marker message here that made your rviz crash? I wonder why the values trigger the assertion if they are not large.

christian-rauch, 2019-08-06:
I think my issue was actually not caused by the marker itself.
Can you check if https://github.com/ros-visualization/rviz/pull/1396 resolves your issue?

christian-rauch, 2019-08-06:
My issue was related to some missing transformations that resulted in using uninitialised scale values. Using uninitilised values should be prevented by my PR, but I assume this will not solve your issue.

mvieth, 2019-08-06:
I will test it, but I also don't think this will solve my issue

mvieth, 2019-08-07:
No, it is not the solution for my issue

christian-rauch, 2019-08-07:
@mvieth If you send me a small log with the faulty markers, I can have a look and integrate a fix into my PR.

mvieth, 2019-08-07:
@christian-rauch Thank you for your offer, I already created a PR before I read your last comment. Would be great if you have a look at it and let me know what you think.

christian-rauch, 2019-08-07:
I see that there is already a dedicated function to test floats in `rviz/validate_floats.h`. Looks alright to me.

rhaschke, 2019-08-13:
Fixed via #1400.

closed 2019-08-13 by rhaschke