ros-visualization/rviz #1414: Melodic release

rhaschke, 2019-09-01:
I would like to prepare a new release into Melodic asap. However, because there are many new commits, I would like to get some feedback from the community first to resolve potential regressions. Particularly, @wjwwood, @simonschmeisser, @VictorLamoine, @v4hn could you please use the `melodic-devel` branch on a daily basis and report (new) issues?

* [feature] RobotModelDisplay: Poll robot_description at 1Hz until found
* [feature] Use assimp to load STL files (#1365)
* [feature] Added frame-aligned view controller (#1405)
* [feature] Systematically check parameters of incoming marker messages (#1275, #1400)
* [feature] MarkerDisplay: clear statuses on disable()
* [feature] Increase zoom range in orbit view controllers (#1373)
* [feature] Add shortcut "C" to the Publish Point tool (#1321)
* [fix] Fixed several issues with Camera and Image display (#1409)
* [fix] Fixed several issues with view controllers, particularly FPS (#1407)
* [fix] Fixed (small) memory leak in TFDisplay (#1408)
* [fix] Fixed memory leak in MapDisplay (#1406)
* [fix] Properly resize RenderWidget on high-DPI displays (#1263)
* [fix] Fixed opacity of Collada meshes (#1387)
* [fix] Enable non-ascii chars for MovableText (#1374)
* [fix] Ignore nan and inf values when normalizing images (#1378)
* [fix] MovableText: consider full translation vector (#1375)
* [maintanence] ToolManager: simplify key handling code
* [maintanence] Move sip bindings' build directory into the build space. (#1360)
* [maintanence] Use std random generators for portability on Windows (#1356)
* [api] public CovarianceVisual (#1410)
* [api] Properly override FailedDisplay::save (#1402)
* [api] Removed extra wrapper MarkerArrayDisplay::handleMarkerArray (#1401)

v4hn, 2019-09-02:
Thanks for asking.
Running it, I noticed a regression introduced by #1275 . See #1415 .

Also, I've been fighting a lot with the bugs mentioned in #1409 recently.
I'll test the latest branch with our setup later today.

simonschmeisser, 2019-09-02:
There seems to be quite some visual regression. Before:
[image]

after:
[image]

Those are all stls with a color set from urdf.

Also I see a new crash which I'll debug now. We should hold back the release a bit

rhaschke, 2019-09-02:
@simonschmeisser, can you please verify whether the visual regression is due to your #1365?
The only other commit, I can think of, that affects rendering is #1387. If those are not the culprit, could you please do a git bisect. Alternatively, please provide your URDF (and meshes), so that I can do it.
Thanks a lot.

simonschmeisser, 2019-09-02:
it worked after #1387 , will test further but first fix the crash

simonschmeisser, 2019-09-02:
So the crash is here: https://github.com/ros-visualization/rviz/blob/325e74cf3dc1bdd1db859beae88db85c926948c9/src/rviz/default_plugin/markers/text_view_facing_marker.cpp#L86 with `text_` being `0`

rhaschke, 2019-09-02:
@v4hn, this issue might be related to your #1275 as well, not defining the text marker anymore if there is some issue (e.g. an empty text?). @simonschmeisser, can you please provide a yaml or rosbag record of your marker message?

simonschmeisser, 2019-09-02:
`rostopic echo /scene_objects/update` gives me

```
server_id: "/move_group"
seq_num: 1
type: 0
markers: []
poses: []
erases: []
---
server_id: "/move_group"
seq_num: 1
type: 0
markers: []
poses: []
erases: []
---
server_id: "/move_group"
seq_num: 2
type: 1
markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "world_frame"
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    name: ''
    description: ''
    scale: 1.0
    menu_entries: []
    controls: 
      - 
        name: ''
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 0.0
        orientation_mode: 0
        interaction_mode: 0
        always_visible: True
        markers: 
          - 
            header: 
              seq: 0
              stamp: 
                secs: 0
                nsecs:         0
              frame_id: ''
            ns: ''
            id: 0
            type: 1
            action: 0
            pose: 
              position: 
                x: 0.0
                y: 0.0
                z: 0.0
[... 279 lines cut ...]
        independent_marker_orientation: False
        description: ''
      - 
        name: "rotate_z"
        orientation: 
          x: 0.0
          y: -1.0
          z: 0.0
          w: 1.0
        orientation_mode: 0
        interaction_mode: 5
        always_visible: False
        markers: []
        independent_marker_orientation: False
        description: ''
poses: []
erases: []
---
server_id: "/move_group"
seq_num: 5
type: 0
markers: []
poses: []
erases: []
```

simonschmeisser, 2019-09-02:
@rhaschke it is indeed due to my commit #1365 , I found that the `stl_loader` sets
https://github.com/ros-visualization/rviz/blob/6135ee28067eb40096808329d6a6ae84c7da054a/src/rviz/ogre_helpers/stl_loader.cpp#L236

will now check if something similar gets set with the assimp code path

simonschmeisser, 2019-09-02:
ok, changing the following https://github.com/ros-visualization/rviz/blob/325e74cf3dc1bdd1db859beae88db85c926948c9/src/rviz/mesh_loader.cpp#L379

to

`submesh->setMaterialName("BaseWhiteNoLighting", "rviz");`

fixes STL coloring, this should of course be dependent on some property of the mesh, not sure which

simonschmeisser, 2019-09-02:
as a test-case you could use https://github.com/ros-industrial/fanuc  and then run

`roslaunch fanuc_lrmate200ib_moveit_config demo.launch`

it should show a bright and friendly orange instead of the gloom dark one

simonschmeisser, 2019-09-02:
about the crash, it appears that `TextViewFacingMarker` discards messages containing purely whitespace:
https://github.com/ros-visualization/rviz/blame/melodic-devel/src/rviz/default_plugin/markers/text_view_facing_marker.cpp#L60
which was introduced in #1275 however there is no check for `nullptr` in https://github.com/ros-visualization/rviz/blame/melodic-devel/src/rviz/default_plugin/markers/text_view_facing_marker.cpp#L88

Also the name `onNewMessage` implies that this might be used to update a marker? So changing the caption from non-whitespace to whitespace would get discarded here. I don't think that's correct either?

rhaschke, 2019-09-02:
> `submesh->setMaterialName("BaseWhiteNoLighting", "rviz");`
fixes STL coloring. This should, of course, depend on some property of the mesh. Not sure which.

Would be great, of course, if you could figure out this.

Regarding the crash, @v4hn introduced this white-space-only check in #1275, because they noticed Ogre would crash with such text. Thus, setting an empty text should delete the marker on an update.
@v4hn, could you have a look?

simonschmeisser, 2019-09-03:
> > `submesh->setMaterialName("BaseWhiteNoLighting", "rviz");`
> > fixes STL coloring. This should, of course, depend on some property of the mesh. Not sure which.
> 
> Would be great, of course, if you could figure out this.

I'll try to investigate sometime later this week what's the difference between a collada mesh and a stl mesh after it has been parsed by assimp. Alternatively we could copy the old behavior and set the material based on file extension

simonschmeisser, 2019-09-06:
There are two parts to the coloring issue:

1. The coloring issues seem to be a known and fixed issue in assimp 4.1.0 (as used on ubuntu bionic): https://github.com/assimp/assimp/pull/2563  a workaround cited there replaces pure white ambient light with pure black one: https://github.com/mosra/magnum-plugins/commit/6444d543d943a7c60e2263f6417517b0b699d231#diff-bd6ab8f40c2db2c93f2d75de9e4997ed

2. We only apply the urdf color to links that have "BaseWhite" or "BaseWhiteNoLighting" as material name: https://github.com/ros-visualization/rviz/blob/b1b5a96b85c1bdfcb0657ce6cd846ca33040f9af/src/rviz/robot/robot_link.cpp#L631

So it appears we need to do some magic based on file extensions?

rhaschke, 2019-09-20:
Release prepared: https://github.com/ros/rosdistro/pull/22320

closed 2019-09-20 by rhaschke