ros-visualization/rviz #1418: InteractiveMarker: do not crash on unknown marker type

simonschmeisser, 2019-09-04:
https://github.com/ros-visualization/rviz/blob/325e74cf3dc1bdd1db859beae88db85c926948c9/src/rviz/default_plugin/interactive_markers/interactive_marker_control.cpp#L93

should add a continue

v4hn, 2019-09-04:
This is true, but also `nan` values in arbitrary fields, or otherwise incorrectly specified marker properties might entail crashes or exceptions.

It probably makes sense to `checkMarkerMsg` each entry and skip it (+ error message) when checking fails.

simonschmeisser, 2019-09-04:
> It probably makes sense to `checkMarkerMsg` each entry and skip it (+ error message) when checking fails.

you're welcome :wink:

rhaschke, 2019-09-08:
Fixed via b1fc2e6214341b69d8d6370479577d9dd90505c3.

closed 2019-09-08 by rhaschke