ros-visualization/rviz #1420: `rviz -d xyz` could look for `~/.rviz/xyz.rviz` if xyz does not exist? [feature request]

v4hn, 2019-09-05:
This is just a minor potential usability improvement I came up with recently.
I use `rviz -d /path/to/file.rviz` quite a lot these days, working on a machine with multiple users of rviz.

Of course, this can be automated in a number of ways already, creating a shell alias, scripts or setup a launch file, but why not support the use-case in RViz directly?

Would it be a welcome feature in RViz to complete `rviz -d xyz` to `rviz -d $HOME/.rviz/xyz.rviz` if the original `xyz` is no valid file?

rhaschke, 2019-09-06:
Why not. If you have it prepared already, please file a PR.

v4hn, 2019-09-06:
We work in different ways. :grinning: 
I tend to ask whether it makes sense to write a patch for upstream *before* I invest the time to do it. ;)
I just wrote a patch and filed it as #1422

rhaschke, 2019-09-08:
Fixed via #1422.

closed 2019-09-08 by rhaschke