ros-visualization/rviz #1426: multipel 'RobotModel' cannot have different alpha

christian-rauch, 2019-09-20:
I have two `RobotModel` (A and B) added to RViz, which show the same robot model at different configurations (A, B). Both widget have separate settings for the Alpha value. However, if I change the alpha for one of the `RobotModel` (A), the other one changes (B) too:
[image]

The alpha value is not updated in the parameter overview of the other widget B, but it seems as if both models share the same visual properties.

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
rviz version 1.13.3
compiled against Qt version 5.9.5
compiled against OGRE version 1.9.0 (Ghadamon)
```

rhaschke, 2019-09-21:
Duplicate of #1385. Fixed in #1387, which was released into 1.13.4.

closed 2019-09-21 by rhaschke

christian-rauch, 2019-09-21:
Thanks for fixing this. I didn't notice that duplicate bug since I only searched in the open issues.
Looking forward to a binary release :-)