ros-visualization/rviz #1431: Cannot mix incompatible Qt versions

eugval, 2019-09-23:
Following the installation instructions form [here](http://wiki.ros.org/rviz/UserGuide#Install_or_build_rviz) , I installed rviz and run

``` rosrun rviz rviz```

And I get the following output:
[image]

It seems that I need to upgrate Qt to version 5.13 from 5.95, but even after following the instructions from [here](https://unix.stackexchange.com/questions/481063/ubuntu-18-04-1-lts-x64-how-to-update-qt-5-10-from-qt-5-9-5):
- Install the newest Qt Version from the Qt website
- Update the qtchooser ```default.conf``` to point to the newly installed version of Qt,

the result is the same and when I run rviz the output is the same.

I am quite new to ROS and Rviz, any help in how to fix this would be much appreciated!

I am running the following:
* OS Version:  Ubuntu 18.04
* ROS Distro: Melodic
 *  ```qtchooser -l``` gives 
[image]
* ```qmake --version points to the (correct) newly installed version of Qt ```
[image]

rhaschke, 2019-09-23:
When you installed rviz from binary packages (via `sudo apt-get install ros-melodic-rviz`), this rviz version is bound to the default Qt version shipped with Ubuntu (which is 5.9.5).
If you want to use your private Qt version, you need to build rviz from source. But actually, I don't recommend this, if you didn't manage the above.

closed 2019-09-23 by rhaschke

eugval, 2019-09-24:
Thanks!

Any ideas why it gave me the incompatible Qt library error message then when I just installed through apt or how could I fix it?

I thought because it was expecting a later version of Qt than the Ubuntu default so I installed 5.13.1, but as you say this was not the issue?

rhaschke, 2019-09-24:
Probably you had another Qt version installed in your system and included via `LD_LIBRARY_PATH`.
On a clean Ubuntu installation, you won't observe such issues.

eugval, 2019-09-24:
Fantastic! That was it, thank you. 

For reference :  [VRep](http://www.coppeliarobotics.com/) seemed to be the issue. When I cleared the LD_LIBRARY_PATH of the path to v-rep, it started working, so it probably uses its own version of Qt, and points to it via LD_LIBRARY_PATH.

yquantao, 2021-03-30:
> Fantastic! That was it, thank you.
> 
> For reference : [VRep](http://www.coppeliarobotics.com/) seemed to be the issue. When I cleared the LD_LIBRARY_PATH of the path to v-rep, it started working, so it probably uses its own version of Qt, and points to it via LD_LIBRARY_PATH.

I met the same problem, how did you clear the LD_LIBRARY_PATH? I commented out it in .bashrc, but it does not work.

rhaschke, 2021-03-30:
When you updated .bashrc, you need to create a new terminal first to get the updated environment.

RahulraaghavA1308, 2022-09-05:
> Fantastic! That was it, thank you.
> 
> For reference : [VRep](http://www.coppeliarobotics.com/) seemed to be the issue. When I cleared the LD_LIBRARY_PATH of the path to v-rep, it started working, so it probably uses its own version of Qt, and points to it via LD_LIBRARY_PATH.

Yeah Thank you so much. this worked for me. I commented those lines in the .bashrc file and the issue was solve

lgzid, 2023-08-10:
@yquantao how can i use vrep and rviz together

rhaschke, 2023-08-10:
@lgzid, simply don't set a default `LD_LIBRARY_PATH` (aiming for a one-fits-all configuration, which doesn't work) but configure it depending on the application(s) to start. Sourcing ROS' setup.bash files automatically configures (actually augments) the `LD_LIBRARY_PATH` with all stuff required for ROS. If you have a similar script for VREP your done.

rhaschke, 2023-08-10:
You should remove the LD_LIBRARY_PATH setting from your `.bashrc`. Just keep:
```
export COPPELIASIM_ROOT=/home/***/software/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04
export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT
```
Create a separate script `vrep.sh` comprising:
`LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$COPPELIASIM_ROOT" vrep`

lgzid, 2023-08-10:
@rhaschke 
Thank you for your reply！
Actually, I need to use rviz, vrep, and [pyrep](https://github.com/stepjam/PyRep) to complete my work.

I found the problem in the third line：

`export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT`

When I cleared it, rviz and vrep worked normally.But the [pyrep](https://github.com/stepjam/PyRep) doesn't work because it can't find the path to QT.

I just tried your approach, but it still hasn't worked.
This is my modified file:
.bashrc:
`export COPPELIASIM_ROOT=/home/***/software/CoppeliaSim_Edu_V4_1_0_Ubuntu20_04

export QT_QPA_PLATFORM_PLUGIN_PATH=$COPPELIASIM_ROOT`

coppeliasim.sh:
`#!/bin/bash

thisscript="$0"
while [ -L "$thisscript" ]; do
        thisscript="`readlink "$thisscript"`"
done

dirname=`dirname "$thisscript"`
if [ $dirname = "." ]; then
        dirname="$PWD"
fi

appname="`basename "$thisscript" | sed 's,\.sh$,,'`"

PARAMETERS=( ${@} )

FILE_PATTERN1='*ttt'
FILE_PATTERN2='*ttm'
FILE_PATTERN3='*simscene.xml'
FILE_PATTERN4='*simmodel.xml'
for i in `seq 0 $(( ${#PARAMETERS[@]} -1 ))`
do
  if [ -f "${PARAMETERS[$i]}" ] && ( [[ "${PARAMETERS[$i]}" == $FILE_PATTERN1 ]] || [[ "${PARAMETERS[$i]}" == $FILE_PATTERN2 ]] || [[ "${PARAMETERS[$i]}" == $FILE_PATTERN3 ]] || [[ "${PARAMETERS[$i]}" == $FILE_PATTERN4 ]] )
  then
    if [ -f "$PWD/${PARAMETERS[$i]}" ]
    then
      PARAMETERS[$i]="$PWD/${PARAMETERS[$i]}"
    fi
  fi
done

# LD_LIBRARY_PATH=$dirname:$LD_LIBRARY_PATH (aboriginal)
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$COPPELIASIM_ROOT
export LD_LIBRARY_PATH

"$dirname/$appname" "${PARAMETERS[@]}"`

zhaianbang, 2024-12-17:
> Fantastic! That was it, thank you.
> 
> For reference : [VRep](http://www.coppeliarobotics.com/) seemed to be the issue. When I cleared the LD_LIBRARY_PATH of the path to v-rep, it started working, so it probably uses its own version of Qt, and points to it via LD_LIBRARY_PATH.

it works for me.