ros-visualization/rviz #1432: inactive RobotState shows wrong robot state after reset

christian-rauch, 2019-09-24:
Steps:
1. add a `RobotState` widget
2. deactivate the widget (uncheck)
3. press the `Reset` button in the bottom left corner
4. the links should not be shown because the widget is inactive, but they are shown with the identity transform

Example image with the Kinova Jaco model:
[image]

The widget goes back to the expected behaviour, when toggling the widget on/off.

### Your environment
Ubuntu 18.04 / RViz version 1.13.4 (melodic).

rhaschke, 2019-09-24:
`RobotState` is a MoveIt plugin. Could you file this issue please there?

closed 2019-09-24 by rhaschke

christian-rauch, 2019-09-24:
@rhaschke Is it possible for you to move this issue directly to the MoveIt repository? See: https://help.github.com/en/articles/transferring-an-issue-to-another-repository

rhaschke, 2019-09-24:
Didn't know about this possibility yet. However, looks like transferring is limited to the organization scope. I cannot choose the moveit repository.

christian-rauch, 2019-09-24:
Issue moved to https://github.com/ros-planning/moveit/issues/1674

I see, RViz and MoveIt are in different organisations. I read "You can only transfer issues between repositories owned by the same user or organization account." as if you can also transfer the issues if a user is an administrator at both organisations.