ros-visualization/rviz #1438: Segmentation fault when running rviz on wsl in Windows10

Beniko95J, 2019-10-22:
Describe your issue here and explain how to reproduce it:

I am trying to use ROS on wsl in Windows10 while I met the segmentation fault (core dumped) error when running rviz. I have searched a lot about the segmentation fault when running rviz but I have not found the solution for my case.

### Your environment
* OS Version: wsl(Ubuntu 18.04) on Windows10
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
     ~ rosrun rviz rviz
     QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-benik'
     [ INFO] [1571763679.870059200]: rviz version 1.13.4
     [ INFO] [1571763679.871324300]: compiled against Qt version 5.9.5
     [ INFO] [1571763679.872056300]: compiled against OGRE version 1.9.0 (Ghadamon)
     [ INFO] [1571763679.897642600]: Forcing OpenGl version 0.
     [1]    21077 segmentation fault (core dumped)  rosrun rviz rviz
    ```
I use VcXsrv as X11 server and glxgears and other ROS gui programs(like rqt_graph) works fine in current settings.

Here are some other console logs that I think may be helpful:

```
~ glxinfo | grep "version string"
server glx version string: 1.4
client glx version string: 1.4
OpenGL version string: 1.4 (2.1 Mesa 19.2.0-devel (git-cdf42f5eaa))
~ env | grep LIBGL
LIBGL_ALWAYS_INDIRECT=1
LIBGL_ALWAYS_SOFTWARE=1
```

The gdb backtrace output is as follows:

```
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffc8b978af in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#2  0x00007fffc8b715ca in Ogre::GLStateCacheManager::switchContext(long) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#3  0x00007fffc8b64c9d in Ogre::GLRenderSystem::initialiseContext(Ogre::RenderWindow*) ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007fffc8b6c754 in Ogre::GLRenderSystem::_createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*)
    () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffffa6f3347 in Ogre::Root::createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007fffff0fc47b in rviz::RenderSystem::tryMakeRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const*, int) ()
   from /opt/ros/melodic/lib/librviz.so
#7  0x00007fffff0fe142 in rviz::RenderSystem::makeRenderWindow(unsigned long, unsigned int, unsigned int, double) ()
   from /opt/ros/melodic/lib/librviz.so
#8  0x00007fffff100804 in rviz::RenderSystem::RenderSystem() () from /opt/ros/melodic/lib/librviz.so
#9  0x00007fffff100965 in rviz::RenderSystem::get() () from /opt/ros/melodic/lib/librviz.so
#10 0x00007fffff0fb491 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) ()
   from /opt/ros/melodic/lib/librviz.so
#11 0x00007fffff122e29 in rviz::RenderPanel::RenderPanel(QWidget*) () from /opt/ros/melodic/lib/librviz.so
#12 0x00007fffff16cdeb in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/melodic/lib/librviz.so
#13 0x00007fffff176ed6 in rviz::VisualizerApp::init(int, char**) () from /opt/ros/melodic/lib/librviz.so
#14 0x0000000008000dc0 in main ()
```

rhaschke, 2019-10-22:
Unfortunately, I cannot help with Windows, but according to the backtrace, the problem is related to creating the GL context in Ogre. As Ogre 1.9.0 shipped with Ubuntu is rather old, you might want to try building a more recent Ogre version from source and then building rviz' noetic-devel branch, which is compatible with newer Ogre versions.

KadynCBR, 2020-01-28:
Had this issue, for the next traveler, what solved it for me was making unsetting the environment variable like so:
 `LIBGL_ALWAYS_INDIRECT=`

deeloon, 2020-02-22:
> Had this issue, for the next traveler, what solved it for me was making unsetting the environment variable like so:
> `LIBGL_ALWAYS_INDIRECT=`

Thank you, this solved my issue

ikramooe, 2020-03-22:
@

> Had this issue, for the next traveler, what solved it for me was making unsetting the environment variable like so:
> `LIBGL_ALWAYS_INDIRECT=`

solved my problem thank you

streamB, 2020-04-27:
> Had this issue, for the next traveler, what solved it for me was making unsetting the environment variable like so:
> `LIBGL_ALWAYS_INDIRECT=`

works for me, thank you very much!

closed 2020-04-27 by rhaschke

kosbeam, 2021-06-23:
To solve this bug in wsl ubuntu 20.04 on windows 10 use this command
$ echo "export DISPLAY=:0" >> ~/.bashrc
$ source ~/.bashrc
$ echo 'export LIBGL_ALWAYS_INDIRECT=0' >> ~/.bashrc

smhadisadati, 2021-11-11:
> To solve this bug in wsl ubuntu 20.04 on windows 10 use this command $ echo "export DISPLAY=:0" >> ~/.bashrc $ source ~/.bashrc $ echo 'export LIBGL_ALWAYS_INDIRECT=0' >> ~/.bashrc

Solved for me, thanks!

jwwang0-0, 2024-01-18:
I still have the segmentation error even though I followed the above setting. Any other solutions?

rhaschke, 2024-01-25:
No, no more ideas for now. Sorry.