ros-visualization/rviz #1439: Rviz crashes when visualization marker contains nan in values

SwiftGust, 2019-10-25:
Describe your issue here and explain how to reproduce it.

[Local_Planner_SITL-Land-Rviz-kill.zip](https://github.com/ros-visualization/rviz/files/3770170/Local_Planner_SITL-Land-Rviz-kill.zip)

Playback uploaded rosbag file and visualize /path_waypoint topic (type - visualization::Marker) alone
when it contains nan in the points value, 
Rviz crashes with following message
```
rviz: /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
Aborted (core dumped)
```

where the topic contents are
```
---
header: 
  seq: 387
  stamp: 
    secs: 39
    nsecs: 464000000
  frame_id: "local_origin"
ns: ''
id: 387
type: 4
action: 0
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
scale: 
  x: 0.02
  y: 0.0
  z: 0.0
color: 
  r: 1.0
  g: 0.0
  b: 0.0
  a: 1.0
lifetime: 
  secs: 0
  nsecs:         0
frame_locked: False
points: 
  - 
    x: -0.0114045487717
    y: -0.039681263268
    z: nan
  - 
    x: -0.0114045487717
    y: -0.039681263268
    z: nan
colors: []
text: ''
mesh_resource: ''
mesh_use_embedded_materials: False
---

```

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
   [ INFO] [1571970533.066781595]: rviz version 1.13.5
   [ INFO] [1571970533.066812126]: compiled against Qt version 5.9.5
   [ INFO] [1571970533.066817226]: compiled against OGRE version 1.9.0 (Ghadamon)
   [ INFO] [1571970533.074023763]: Forcing OpenGl version 0.
   [ INFO] [1571970533.657846107]: Stereo is NOT SUPPORTED
   [ INFO] [1571970533.657907318]: OpenGl version: 4.6 (GLSL 4.6).
    ```
* If source build, which git commit?
commit e49a5448f2a763dcba2963a7e87e627b6db6d1b6

rhaschke, 2019-10-25:
It's interesting that a lot of similar issues (with malformed messages) popped up recently.
I filed a fix in #1440. Could you please verify this for your use case?

SwiftGust, 2019-10-25:
@rhaschke It is working, I'm getting warning messages instead of rviz crash

```
[ WARN] [1572019736.235473679, 146.560000000]: Marker '/1458': invalid point[0] (-0.03, -0.14, nan)
[ WARN] [1572019736.235518609, 146.560000000]: Marker '/1458': invalid point[1] (-0.03, -0.14, nan)
[ WARN] [1572019736.332717967, 146.656000000]: Marker '/1459': invalid point[0] (-0.03, -0.14, nan)
[ WARN] [1572019736.332755887, 146.656000000]: Marker '/1459': invalid point[1] (-0.03, -0.14, nan)
[ WARN] [1572019736.428963777, 146.752000000]: Marker '/1460': invalid point[0] (-0.03, -0.14, nan)
[ WARN] [1572019736.429001977, 146.752000000]: Marker '/1460': invalid point[1] (-0.03, -0.14, nan)
[ WARN] [1572019736.523704624, 146.848000000]: Marker '/1461': invalid point[0] (-0.03, -0.14, nan)
[ WARN] [1572019736.523748744, 146.848000000]: Marker '/1461': invalid point[1] (-0.03, -0.14, nan)
[ WARN] [1572019736.619853312, 146.944000000]: Marker '/1462': invalid point[0] (-0.03, -0.14, nan)
[ WARN] [1572019736.619893252, 146.944000000]: Marker '/1462': invalid point[1] (-0.03, -0.14, nan)
```

PR #1440 (2019-10-27): LineStripMarker: ignore invalid points
Fix #1439.
closed 2019-10-27 by rhaschke