ros-visualization/rviz #1441: Segmentation fault when measuring something in ROS

TheCynosure, 2019-11-08:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz:
    ```
[ INFO] [1573250272.926426956]: rviz version 1.13.4
[ INFO] [1573250272.926474597]: compiled against Qt version 5.9.5
[ INFO] [1573250272.926488049]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1573250272.929100038]: Forcing OpenGl version 0.
[ INFO] [1573250273.499971391]: Stereo is NOT SUPPORTED
[ INFO] [1573250273.500043323]: OpenGl version: 4.6 (GLSL 4.6).

    ```
I am displaying a large pointcloud that is constantly moving (A slam implementation). I am then going and trying to measure the distance between two points and this is causing a segmentation fault. Specifically I click the measure tool, then click on the map is when the program crashes.
Here is the stacktrace that I collected from GDB:

```
#0  0x00000000400b3d0a in  ()
#1  0x00007fffb7319184 in  () at /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.430.50
#2  0x00007fffb6e9284d in  () at /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.430.50
#3  0x00007fffb51f691a in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) () at /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007ffff31c8ba9 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#5  0x00007ffff316b0a9 in Ogre::QueuedRenderableCollection::acceptVisitorDescending(Ogre::QueuedRenderableVisitor*) const () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff316b191 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff31b51fc in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff31b502e in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff31cabfb in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff2fd9d2a in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff318647a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff318628b in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff318600e in Ogre::RenderTarget::updateImpl() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff3186a34 in Ogre::RenderTarget::update(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff316f4df in Ogre::RenderSystem::_updateAllRenderTargets(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff31aca4a in Ogre::Root::_updateAllRenderTargets() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff31acb30 in Ogre::Root::renderOneFrame() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff7b4bde8 in rviz::VisualizationManager::onUpdate() () at /opt/ros/melodic/lib/librviz.so
#19 0x00007ffff6cc1645 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff6cce0b7 in QTimer::timeout(QTimer::QPrivateSignal) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff6cce418 in QTimer::timerEvent(QTimerEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff6cc216b in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff72a783c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff72af104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff6c929c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff6ceae1e in QTimerInfoList::activateTimers() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff6ceb5e1 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff0023417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff0023650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff00236dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff6ceb97f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff6c909fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff6c99aa4 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x0000555555554dc9 in main ()
```

rhaschke, 2019-11-10:
I cannot reproduce this issue. Could you please provide a minimal example robustly showcasing this segfault? For example, provide a small program to publish a moving point cloud or provide a rosbag file of your point cloud topic.
Does the segfault occur on/after? the first or the second click?

TheCynosure, 2019-11-12:
Let me see if I am able to publish the bag file on here.

closed 2019-11-25 by TheCynosure

rhaschke, 2019-11-25:
Because you closed this issue, I guess you were not able to reproduce the issue?

TheCynosure, 2020-02-07:
I have been able to zoom in very close to large pointclouds and try and take measurements causing crashing many times but I can't seem to reproduce it / come up with a scenario where it always keeps occuring.