ros-visualization/rviz #1442: [melodic|Fedora 31] Rviz segfault under wayland but work under X

MalcolmMielle, 2019-11-14:
Hi!

It's impossible to start rviz under Wayland but it starts fine under X. Under Wayland, rviz segfaults after complaining that Ogre as the wrong screen/server:

```
[ WARN] [1573737603.322616552]: OGRE EXCEPTION(3:RenderingAPIException): Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at /builddir/build/BUILD/ogre-1.9.0/RenderSystems/GL/src/GLX/OgreGLXWindow.cpp (line 240)
rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException): Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at /builddir/build/BUILD/ogre-1.9.0/RenderSystems/GL/src/GLX/OgreGLXWindow.cpp (line 240)
```
I have tried `export LIBGL_ALWAYS_SOFTWARE=1` but the result is the same.

### Your environment
* OS Version: Fedora 31
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1573737603.105062109]: rviz version 1.13.5
    [ INFO] [1573737603.105137221]: compiled against Qt version 5.12.5
    [ INFO] [1573737603.105158824]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1573737603.114214707]: Forcing OpenGl version 0.
    [ INFO] [1573737603.314984160]: Stereo is NOT SUPPORTED
    [ INFO] [1573737603.315102243]: OpenGl version: 3 (GLSL 1.3).

    ```
* If source build, which git commit?

I have used the source install method of the wiki. So I'm at whichever commit this links to.

rhaschke, 2019-11-14:
This seems to be a Wayland issue, not providing full X compatibility. Did you file an issue there?

closed 2019-11-14 by rhaschke

MalcolmMielle, 2019-11-14:
Ok I've found a temporary solution by forcing the use of X11 on Wayland using `QT_QPA_PLATFORM=xcb ./rviz`.

I'm not sure whether Wayland is supposed to provide full X compatibility (some program notoriously won't work on Wayland but work on X) or if Rviz needs to be updated for wayland. If the consesnsus is that it's a Wayland problem, I'll go report the error there.

rhaschke, 2019-11-14:
As indicated by the error, the culprit is OGRE, which doesn't support Wayland yet:
https://forums.ogre3d.org/viewtopic.php?t=93084#p537787

rhaschke, 2019-11-14:
> I'm not sure whether Wayland is supposed to provide full X compatibility

As far as I know, wayland allows to run an X11 server and thus provide the platform to run any X11 client program - exactly what you did with `QT_QPA_PLATFORM=xcb`.

However, to natively support wayland, besides relying on corresponding OGRE support, rviz will also require to replace some native X11 calls. See here for some hints:
https://blog.martin-graesslin.com/blog/2015/07/porting-qt-applications-to-wayland/
https://wayland.freedesktop.org/qt5.html

ionutnechita-intel, 2024-02-06:
Hello,

This is happened when compiling with colcon using rolling branch.

RViz version 13.3.1 (ROS2 rolling)
Qt version 5.15.10
OGRE version 1.12.10 (Rhagorthua)