ros-visualization/rviz #1444: Marker::POINTS ignoring attributes

ligazetom, 2019-11-19:
Hey everyone.

I'm visualizing processed point cloud data using Marker::POINTS (Not sure if right approach because of PointCloud2, but eh).

The thing is, no matter what attribute I set: scale, color (really can distinguish only between these two), they are ignored by rviz. The easiest way to reproduce is to change one of tutorials: http://wiki.ros.org/rviz/Tutorials/Markers%3A%20Basic%20Shapes

`    #include <ros/ros.h>
    #include <visualization_msgs/Marker.h>

    int main( int argc, char** argv )
    {
      ros::init(argc, argv, "basic_shapes");
      ros::NodeHandle n;
      ros::Rate r(1);
      ros::Publisher marker_pub = n.advertise<visualization_msgs::Marker>("visualization_marker", 1);

      // Set our initial shape type to be a cube
      uint32_t shape = visualization_msgs::Marker::CUBE;

      while (ros::ok())
      {
        visualization_msgs::Marker marker;
        // Set the frame ID and timestamp.  See the TF tutorials for information on these.
        marker.header.frame_id = "/my_frame";
        marker.header.stamp = ros::Time::now();

        // Set the namespace and id for this marker.  This serves to create a unique ID
        // Any marker sent with the same namespace and id will overwrite the old one
        marker.ns = "basic_shapes";
        marker.id = 0;

        // Set the marker type.  Initially this is CUBE, and cycles between that and SPHERE, ARROW, and CYLINDER
        marker.type = shape;

        // Set the marker action.  Options are ADD, DELETE, and new in ROS Indigo: 3 (DELETEALL)
        marker.action = visualization_msgs::Marker::ADD;

        // Set the pose of the marker.  This is a full 6DOF pose relative to the frame/time specified in the header
        marker.pose.position.x = 0;
        marker.pose.position.y = 0;
        marker.pose.position.z = 0;
        marker.pose.orientation.x = 0.0;
        marker.pose.orientation.y = 0.0;
        marker.pose.orientation.z = 0.0;
        marker.pose.orientation.w = 1.0;

        // Set the scale of the marker -- 1x1x1 here means 1m on a side
        marker.scale.x = 1.0;
        marker.scale.y = 1.0;
        marker.scale.z = 1.0;

        // Set the color -- be sure to set alpha to something non-zero!
        marker.color.r = 0.0f;
        marker.color.g = 1.0f;
        marker.color.b = 0.0f;
        marker.color.a = 1.0;

        marker.lifetime = ros::Duration();

        // Publish the marker
        while (marker_pub.getNumSubscribers() < 1)
        {
          if (!ros::ok())
          {
            return 0;
          }
          ROS_WARN_ONCE("Please create a subscriber to the marker");
          sleep(1);
        }
        marker_pub.publish(marker);

        // Cycle between different shapes
        switch (shape)
        {
        case visualization_msgs::Marker::CUBE:
          shape = visualization_msgs::Marker::SPHERE;
          break;
        case visualization_msgs::Marker::SPHERE:
          shape = visualization_msgs::Marker::POINTS;
          break;
        case visualization_msgs::Marker::POINTS:  
          shape = visualization_msgs::Marker::CYLINDER;
          break;
        case visualization_msgs::Marker::CYLINDER:
          shape = visualization_msgs::Marker::CUBE;
          break;
        }

        r.sleep();
      }
    }`

Where the only change is in **switch statement at the end**, where I switched cylinder for points. Obviously the same can be done in any other tutorial. The points are just always small and gray-ish. Does not matter if you set "color" or "colors" with "ColorRGBA" push_back.

The code I am actually using (class method callback, so it won't run on its own):

`void LidarConeDetection::ProcessData(const sensor_msgs::PointCloud2::ConstPtr& msg)
{
    float const *temp;
    visualization_msgs::Marker points;
    points.type = visualization_msgs::Marker::POINTS;

    points.id = 0;
    points.header.frame_id = "/my_frame";
    points.header.stamp = ros::Time::now();
    points.ns = "points";
    points.action = visualization_msgs::Marker::ADD;
    points.pose.orientation.w = 1.0;
    points.scale.x = 1.0;
    points.scale.y = 1.0;
    points.color.a = 0.;
    points.mesh_use_embedded_materials = false;
  
    std_msgs::ColorRGBA col;

    col.a = 0.;
    col.g = 1.f;
    col.r = 0.;
    col.b = 0.;

    points.points.reserve(msg->width);
    points.colors.reserve(msg->width);

    for (int i = 0; i < msg->width; i++)
    {
        temp = reinterpret_cast<const float*>(&msg->data[i*msg->point_step]);
        geometry_msgs::Point point;

        point.x = *temp;
        point.y = *(temp + 1);
        point.z = *(temp + 2);

        points.points.push_back(point);
        points.colors.push_back(col);
    }
    m_pub.publish(points);    
}`

All done on:
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz - 1.13.5, Qt - 5.9.5, OGRE - 1.9.0, OpenGl - 4.3

I mean it does not matter that much I guess, but I know that is not how it was meant to work.

ligazetom, 2019-11-19:
Also CUBE_LIST is not working as well. Instead of drawing cubes at the positions of points, it draws points themselves and they are white and small.

rhaschke, 2019-11-19:
Thanks for reporting this issue. I'll try to have a look this evening...

ligazetom, 2019-11-19:
> Thanks for reporting this issue. I'll try to have a look this evening...

Now I'll feel bad because of the dots :D. Please, you don't have to do it that fast, LINE_LIST works. There is a way.

rhaschke, 2019-11-19:
Hm. I cannot reproduce your issue. Both the visualization_marker_tutorials' `points_and_lines` demo as well as rviz' `marker_test.cpp` work as expected
[image]
The only difference I see so far is the OpenGL version: mine is 3 only, while yours is 4.3:
```
[ INFO] ros.rviz: rviz version 1.13.5
[ INFO] ros.rviz: compiled against Qt version 5.9.5
[ INFO] ros.rviz: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] ros.rviz: Forcing OpenGl version 0.
[ INFO] ros.rviz: Stereo is NOT SUPPORTED
[ INFO] ros.rviz: OpenGl version: 3 (GLSL 1.3).
```
Where did you get your 4.3 version from? Which graphics card / driver do you use?

ligazetom, 2019-11-20:
Oh god. It was caused by the driver. I had Nouveau originally, which I thought was okay because, well, glxgears were working as well as SOME of the RViz features. Now, when I switched to nvidia 390, the points are showing. So so sorry I've opened this issue.

closed 2019-11-20 by ligazetom

rhaschke, 2019-11-20:
Good that it works for you now. Nevertheless, it should work with nouveau as well (in principle).
Looks like nouveau doesn't use the mesa library, but implements its on OpenGL. Can you confirm this?
Probably you should file an issue with nouveau...

gavanderhoorn, 2019-11-20:
This was also cross-posted to ROS Answers: [RViz POINTS color bug or doing something wrong](https://answers.ros.org/question/337916).

ligazetom, 2019-11-20:
I'd love to help, but this issue seems to be little out of my league. I don't really do much around drivers, but if you meant to try some opengl program with different drivers I could do that.

cedrusx, 2020-04-04:
I have the same issue with an AMD GPU. I have googled a lot and think I have configured the GPU driver properly, but still could not get colored and scaled markers. Could anyone tell me more about this issue? Like which SW package in the whole pipeline is making the decision of ignoring colors and scales...

bxwllzz, 2020-06-30:
The same error occurs when using VMware virtual machine and ROS melodic. The strange thing is that when vmware's accelerated 3D graphics is turned off, this problem will no longer appear, but it will cause extremely slow rendering.

rhaschke, 2020-07-03:
We have seen issues with Mesa OpenGL rendering a lot in the past. Unfortunately, we don't have experts in the team anymore, who could debug those low-level issues. See also #1508.

bxwllzz, 2020-10-28:
By replacing the material files, the problem can be solved in VMware.

```
cd /opt/ros/melodic/share/rviz/ogre_media/materials
mv scripts150 scripts150.bak
cp -r scripts120 scripts150
```