ros-visualization/rviz #1447: terminate called after throwing an instance of 'std::bad_alloc'

yanmengting117, 2019-11-21:
[ INFO] [1574305388.864769319]: rviz version 1.12.17
[ INFO] [1574305388.864812528]: compiled against Qt version 5.5.1
[ INFO] [1574305388.864824468]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1574305392.291733231]: Stereo is NOT SUPPORTED
[ INFO] [1574305392.292059037]: OpenGl version: 3 (GLSL 1.3).
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

rhaschke, 2019-11-21:
This is not very helpful yet. Probably you tried to load a 3rd-party plugin (which one?), which crashed because you are running out of memory. Rerun with `rviz --log-level-debug` to get more info.

yeliang2258, 2020-07-30:
[image]
I have the same problem

rhaschke, 2020-07-30:
Please try the most recent Melodic version of rviz. The log output still isn't very helpful. It would be great if you could compile with debug options (-g -Og) and send the backtrace.

yeliang2258, 2020-07-31:
> Please try the most recent Melodic version of rviz. The log output still isn't very helpful. It would be great if you could compile with debug options (-g -Og) and send the backtrace.

I just ran the 'rosrun rviz rviz' command

rhaschke, 2021-02-05:
Closing due to inactivity. Please feel free to reopen when you can still reproduce on a newly setup system (e.g. docker) and you can provide a backtrace.

closed 2021-02-05 by rhaschke