ros-visualization/rviz #1453: PointCloud: superflous and dubious check

simonschmeisser, 2019-11-27:
While optimizing our code we created some funny point clouds today. Some of them crashed rviz because Ogre could no longer calculate the bounding box of the scene. That remains to be investigated.

However no point ever ended up triggering this check even though we are sending lots of quiet_NaNs (number of points was equal to number of valid points):

https://github.com/ros-visualization/rviz/blob/b9bff9329052764dd7d614688e1310f09da68163/src/rviz/default_plugin/point_cloud_common.cpp#L864

Does anybody know where the first check happens?

rhaschke, 2019-11-27:
Are you using PointCloud2 or PointCloud1? Looks like `transformCloud` is only called for PointCloud2.

simonschmeisser, 2019-11-28:
Yes, PointCloud2 is what we use. The function itself gets called and does the transformation but it is in fact called with only valid points so the check at the end (884ff) is superflous

rhaschke, 2020-11-16:
PointCloud2 immediately filters an incoming message for NaNs:
https://github.com/ros-visualization/rviz/blob/8bd56fb928b255e095424a5f1fb3ff8a5274602e/src%2Frviz%2Fdefault_plugin%2Fpoint_cloud2_display.cpp#L77-L81

However, I think we should keep the additional check because a transformer might have introduced new NANs.

closed 2020-11-16 by rhaschke