ros-visualization/rviz #1455: Segfault when changing view type with tf messages displayed

kartikmohta, 2019-12-11:
Changing view type (for example, from `Orbit` to `ThirdPersonFollower`) when tf messages are displayed in rviz causes a segfault. It seems to be related to recent changes in the `MovableText` class (470fdc819ee8211a4fb0d1c06da19fbbf5330206).

Here is the gdb backtrace:
```
#0  0x00007ffff2fd9857 in Ogre::Camera::getDerivedOrientation() const () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#1  0x00007ffff7ac8420 in rviz::MovableText::getBoundingBox() const () at /opt/ros/melodic/lib/librviz.so
#2  0x00007fffacb2f7ea in Ogre::OctreeNode::_updateBounds() () at /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_OctreeSceneManager.so.1.9.0
#3  0x00007ffff310ae67 in Ogre::Node::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#4  0x00007ffff31e1710 in Ogre::SceneNode::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#5  0x00007ffff310ae67 in Ogre::Node::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff31e1710 in Ogre::SceneNode::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff310ae67 in Ogre::Node::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff31e1710 in Ogre::SceneNode::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff310ae67 in Ogre::Node::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff31e1710 in Ogre::SceneNode::_update(bool, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff31b5129 in Ogre::SceneManager::_updateSceneGraph(Ogre::Camera*) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff31ca830 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff2fd9d2a in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff318647a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff318628b in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff318600e in Ogre::RenderTarget::updateImpl() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff3186a34 in Ogre::RenderTarget::update(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff316f4df in Ogre::RenderSystem::_updateAllRenderTargets(bool) () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff31aca4a in Ogre::Root::_updateAllRenderTargets() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#20 0x00007ffff31acb30 in Ogre::Root::renderOneFrame() () at /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#21 0x00007ffff7b4abe8 in rviz::VisualizationManager::onUpdate() () at /opt/ros/melodic/lib/librviz.so
#22 0x00007ffff6cbd645 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff6cca0b7 in QTimer::timeout(QTimer::QPrivateSignal) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff6cca418 in QTimer::timerEvent(QTimerEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff6cbe16b in QObject::event(QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff72a383c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff72ab104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff6c8e9c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff6ce6e1e in QTimerInfoList::activateTimers() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff6ce75e1 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff0023417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff0023650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff00236dc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#34 0x00007ffff6ce797f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff6c8c9fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007ffff6c95aa4 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x0000555555554dc9 in main ()
```

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
   [ INFO] [1576099938.123090320]: rviz version 1.13.6
   [ INFO] [1576099938.123135278]: compiled against Qt version 5.9.5
   [ INFO] [1576099938.123151468]: compiled against OGRE version 1.9.0 (Ghadamon)
   [ INFO] [1576099938.131705042]: Forcing OpenGl version 0.
   [ INFO] [1576099939.037108248]: Stereo is NOT SUPPORTED
   [ INFO] [1576099939.037238429]: OpenGl version: 4.6 (GLSL 4.6).
    ```
* If source build, which git commit?
  Not source build, this is the melodic binary release.

rhaschke, 2019-12-12:
Thanks for reporting this issue. I can reproduce the problem, but I cannot explain where it comes from: The program counter is reported to be zero in
https://github.com/OGRECave/ogre/blob/fbe7b9edaef1e5368061fed89050092023d61d3d/OgreMain/src/OgreCamera.cpp#L518
However, mpCam is well defined. So, it looks like, the camera's vtable is corrupted for some reason. No idea why.

rhaschke, 2019-12-12:
I found the reason for this issue: MoveableText was still accessing the old ViewController's camera, because `getBoundingBox()` was called before `_notifyCurrentCamera()`.
The best solution, for now, is to revert the offending commit.

kartikmohta, 2019-12-12:
I don't know the exact issue that commit was solving but crashing due to a typical user action seems like bad behavior, so I think that reverting the commit would be best for now.

icolwell-as, 2020-01-03:
A colleague and myself are experiencing the same crashes with the same version of rviz. 
Is the proposed fix reverting this commit?
https://github.com/ros-visualization/rviz/commit/470fdc819ee8211a4fb0d1c06da19fbbf5330206

If so, I'll gladly open a PR to fix this.

kartikmohta, 2020-01-04:
This is already fixed in https://github.com/ros-visualization/rviz/commit/787c40d9f0124b55f66431e0ea43738f82849a65 and part of release v1.13.7.

closed 2020-01-04 by kartikmohta

StanislawAntol, 2020-01-08:
I've recently encountered this issue and am wondering when the [ROS rviz repo](http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-rviz/) will have release v1.13.7.

rhaschke, 2020-01-08:
The released version is already available in the [ROS testing repo](http://wiki.ros.org/action/show/TestingRepository).