ros-visualization/rviz #1458: [melodic][ubuntu18.04] strange visualization

horverno, 2019-12-17:
Basically the issue is, that every time the view changes (e.g. the distance from the scene) the markers are rendered faulty.  
I made a video to illustrate it: https://youtu.be/GoLhq84Qsrg 
To reproduce download the rosbag:
`wget www.sze.hu/~herno/PublicDataAutonomous/leaf-2019-09-12-15-10-46-gps-lidar-zala.bag
`
and watch  `/lane_select_marker` behavior. (In the video I used cube markers only to demonstrate it with larger markers).

### My environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: Melodic
* RViz: 1.13.6 
* Qt 5.9.5
* OGRE 1.9.0
(I also checked in earlier release too in different ubuntu version, but the same issue is present)

Thank you!

rhaschke, 2019-12-17:
I'm not sure what you mean by "the markers are rendered faulty".
Can you become more concrete, please?

rhaschke, 2019-12-17:
I noticed that the linestrip and other markers disappear (probably due to wrong culling) in certain viewing angles of the camera. Looking at your marker messages, I noticed large position offsets. While I agree that this shouldn't pose problems, it obviously does. Shifting all positions to the center, fixes the issue for me. I don't have time to investigate this in detail.

horverno, 2019-12-17:
@rhaschke  Thank you spending time investigating the issue. 
So what you are basically saying is this behavior is related to large coordinates and large position offsets? I think in this case my issue is related to issue #1091

rhaschke, 2019-12-17:
Yes, looks like this is a duplicate of #1091.

horverno, 2019-12-18:
I think in this case the best thing to do is close the issue, but everyone who has something to say or do about this can continue in issue #1091

closed 2019-12-18 by horverno

horverno, 2020-05-13:
note that there is a workaround in issue #502 _(rviz: Large transforms lead to unstable visualizations (ros-pkg ticket #5394))_

horverno, 2020-09-22:
The best workaround i know for this issue is to use a `static_transform_publisher`. As suggested this will "shift" all the positions (or rather the view). This will publish a frame near to the large coordinates, and you can set the RVIZ Fixed frame to this frame. Eg. if your large coordinates are near `x:639770.0 y:5195040.0` and your global frame is called `/map` use this command:

```
rosrun tf static_transform_publisher 639770.0 5195040.0 0.0 0.0 0.0 0.0 /map /map_large 10
```
or this launch:
```
<launch>
<node args="639770.0 5195040.0 0.0 0.0 0.0 0.0 /map /map_large 10" name="map_large_tf_publisher" pkg="tf" type="static_transform_publisher"/>
</launch>
```
after this in Global options set the Fixed frame to `/map_large`  instead of  `/map` . And done :sunglasses: