ros-visualization/rviz #1461: Smooth rendering of spheres

wxmerkt, 2020-01-09:
Hi!
I'd like to add support to display spheres with smoothing. I am using the OGRE built-in `PT_SPHERE` `PrefabType` to create a smooth sphere. However, unlike the `rviz_sphere.mesh` (radius 0.5), these are 100 units in radius by default. Using a scaling of `0.01`, I can get the desired behaviour for the marker array plug-in and robot link. Cf. example below:

### Before
[image]

### After
[image]

### Question
The issue for which I am looking input is how to add this persistently for all plug-ins that may create a `rviz::Shape::Sphere`. If I were to set the scale in `createEntity`, it can easily be overridden by a call to `setScale` from anywhere else. Obviously, I do not want to simply change the convention to avoid breaking plug-ins.

1. Is there a good solution to this I may have overlooked?
2. I believe another solution would be to update the `rviz_sphere.mesh` file. Would this be acceptable?

Many thanks for your input,
Wolfgang

rhaschke, 2020-01-09:
I would suggest to update rviz_sphere.mesh, but maybe @wjwwood as another suggestion.

wjwwood, 2020-01-10:
I don't have a better suggestion. How would you change the mesh file?

wxmerkt, 2020-01-10:
@wjwwood please have a look at #1463 :-)

PR #1463 (2020-01-13): Replace sphere and cylinder with smooth meshes
Fixes #1461 by replacing the existing meshes with meshes that have correct normals and flat shading turned on.

## Changes
- Sphere: Replaced mesh with correct normals as well as shading (may have more vertices, unsure)
- Cylinder: Replaced with mesh with correct normals and shading turned on
- Cone: Replaced with mesh with correct normals and shading turned on

I first used Blender and the blender2ogre exporter to work export a new model of a sphere. However, due to bugs with the export this did not work with the cylinder which is why the cylinde [... cut]
closed 2020-01-13 by rhaschke