ros-visualization/rviz #1464: Rviz crashing with rosparam enable_statistics true (Duration is out of dual 32-bit range)

AndreCid, 2020-01-22:
Using an [Ouster LiDar (OS1-16)](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/README.md), when rosparam _**enable_statistics**_  is _**true**_,  Rviz closes with the following output, when trying to view the IMU (_/os1_cloud_node/imu_) or PoinCloud2 (_/os1_cloud_node/points_) topics: 

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
Aborted (core dumped)
```
This only happens when Rviz is opened after the parameter is changed to true, both in Kinetic and Melodic.

To reproduce the error:
 
```
$ rosbag play rviz_runtime_error.bag
$ rosparam set enable_statistics true
$ rviz # In Rviz, add the IMU or PointCloud nodes 
```

[Bag](https://drive.google.com/file/d/1YKrIHItirvEy10GjKBrODQj2wk8brikk/view) recorded on Ubuntu 16.04 and ROS Kinetic

All Ros Packages are upgraded and Ros workspace was clean rebuilded
    

### Tested environments 
* OS Version: e.g. Ubuntu 16.04 & Ubuntu 18.04
* ROS Distro: Kinetic & Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    
## Ros Kinetic
```
[ INFO] [1579690196.514326580]: rviz version 1.12.17
[ INFO] [1579690196.514376785]: compiled against Qt version 5.5.1
[ INFO] [1579690196.514389307]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1579690196.990473593]: Stereo is NOT SUPPORTED
[ INFO] [1579690196.990538955]: OpenGl version: 4.6 (GLSL 4.6). 
```

## Ros Melodic

```
[ INFO] [1579700455.718782578]: rviz version 1.13.7
[ INFO] [1579700455.718842559]: compiled against Qt version 5.9.5
[ INFO] [1579700455.718857467]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1579700455.734269999]: Forcing OpenGl version 0.
[ INFO] [1579700456.586085000]: Stereo is NOT SUPPORTED
[ INFO] [1579700456.586186504]: OpenGl version: 4.6 (GLSL 4.6).
```

rhaschke, 2020-01-22:
I coarsely remember that this kind of issue with timestamps / durations was related to an ABI incompatibility introduced at some point. Did you rebuild all plugins, particularly the imu plugin?
Which imu plugin you are using?

AndreCid, 2020-01-22:
Yes, It was all rebuilded and all the packages are up to date. The package used to read both IMU and pointcloud data was [Ouster-ROS](https://github.com/ouster-lidar/ouster_example/tree/master/ouster_ros).

rhaschke, 2020-01-22:
I can reproduce the issue. Will have a look...

rhaschke, 2020-01-22:
The issue comes from `ros::StatisticsLogger` with an overflow computing the mean message age.
This issue was fixed in https://github.com/ros/ros_comm/pull/1526, but not yet released.
Please ask there for a new release.

closed 2020-01-22 by rhaschke