ros-visualization/rviz #1465: Laserscan "points" style crashes rviz

Riconec, 2020-01-23:
I am using a virtual machine (vmware) with Kubuntu and I found that rviz config files I created in Ubuntu mate (also VM, but in Virtualbox) earlier works incorrectly: size of points (render style "flat squares") from laserscan messages are ignored, they are all small even if I defined some size.
Changing size won't change anything, but if I try to change flat squares to points style - it will crash:

```[ INFO] [1579774719.942264456]: rviz version 1.13.7
[ INFO] [1579774719.942371352]: compiled against Qt version 5.9.5
[ INFO] [1579774719.942382152]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1579774719.960315568]: Forcing OpenGl version 0.
[ INFO] [1579774721.674050288]: Stereo is NOT SUPPORTED
[ INFO] [1579774721.674331595]: OpenGl version: 3.3 (GLSL 3.3).
kf5.kio.core: Refilling KProtocolInfoFactory cache in the hope to find ""
kf5.kio.core: KSambaShare: Could not find smb.conf!
[ WARN] [1579774775.826586148]: Detected jump back in time of 111.641s. Clearing TF buffer.
../src/gallium/auxiliary/tgsi/tgsi_parse.c:268:tgsi_parse_token: Assertion `0' failed.
Aborted (core dumped)
```

I tried to load Ubuntu mate VM that I was using originally and it works fine on 1.13.3 rviz

So I cloned project and built rviz 1.13.3 (on kubuntu in vmware) to check is that related to my bug and it behaves exatly the same as 1.13.7

### Your environment
Kubuntu 18.04
ROS Melodic, rviz 1.13.7 and 1.13.3 (I have both, one from apt, another compiled)
rviz version 1.13.3
[ INFO] [1579771898.926214393]: compiled against Qt version 5.9.5
[ INFO] [1579771898.926235439]: compiled against OGRE version 1.9.0 (Ghadamon)

Ubuntu mate 18.04
ROS Melodic, rviz 1.13.3

Is that related to VM or opengl used in Ubuntu\Kubuntu? I don't think it is a bug in rviz

rhaschke, 2020-01-23:
Googling the file tgsi_parse.c ends up at [MesaLib](https://github.com/adobe/chromium/blob/master/third_party/mesa/MesaLib/src/gallium/auxiliary/tgsi/tgsi_parse.c). If that worked before in a native Linux installation, I suspect Mesa to be broken when used inside the VM. If so, I cannot help.
Please provide your rviz config for further validation.

Riconec, 2020-01-23:
In both cases I am using VMs, in one case it works, on others - not

rviz config file is here : https://gist.github.com/Riconec/3021e1d9cea06b3299ad483f4d56ac86

rhaschke, 2020-01-23:
Thanks for providing the config. I cannot reproduce the issue with my mesa installation.
Scaling of points correctly adapts with the corresponding settings.
Closing as won't fix in rviz.

closed 2020-01-23 by rhaschke