ros-visualization/rviz #1466: Crash upon entering an invalid topic name (ros::InvalidNameException)

CodeFinder2, 2020-01-29:
When navigating to "2D Nav Goal" > "Topic" in "Tool Properties", RViz crashes immediately with the exception show below when trying to enter invalid characters. IMHO, Rviz should show an error but it shouldn't crash completely. In my case, I entered `/goal[` (accidentally). Note that this happens in every edit fields for "Topics" in the "Tool Properties".

I think the actual content of Rviz (rendered scene) doesn't matter for this.

### Your environment
* OS Version: Ubuntu 18.04 LTS 64 bit
* ROS Distro: Melodic, /rosversion: 1.14.3
* RViz, Qt, OGRE, OpenGl version as printed by rviz: v1.13.7, v5.9.5, v1.9.0 (Ghadamon), v4.6 (GLSL 4.6)

    ```
    terminate called after throwing an instance of 'ros::InvalidNameException'
      what():  Character [(] at element [1] is not valid in Graph Resource Name [/(suppy1/goal].  Valid     characters are a-z, A-Z, 0-9, / and _.
    [rviz-9] process has died [pid 2513, exit code -6, cmd /opt/ros/melodic/lib/rviz/rviz -d /tmp/rviz_config_generator_tmp_saMfAu.rviz __name:=rviz __log:=/.../.ros/log/04fe25dc-429a-11ea-    81b2-e0d55eac4766/rviz-9.log].
    log file: /.../.ros/log/04fe25dc-429a-11ea-81b2-e0d55eac4766/rviz-9*.log
    ```

rhaschke, 2020-01-30:
Thanks for reporting this issue. It's rather easy to catch the exception and just print an error (#1467).
However, I'm not sure how to provide nice feedback at the GUI level about the wrong input.
While displays have a StatusProperty, tools don't have this possibility to report errors. Any suggestions?

CodeFinder2, 2020-01-30:
What about creating a dialog box when the user presses [Enter] after entering invalid input?

Just a message in the log/console might be missed.

PR #1467 (2020-02-08): Catch exceptions when creating publishers in tools
Fix #1466: An invalid ros topic name caused rviz to crash due to the uncaught exception.
closed 2020-02-08 by rhaschke