ros-visualization/rviz #1470: rviz.exe does not work on Win 10

PackChinoros, 2020-02-08:
I've just installed ROS on Win10 and then I installed rviz but I cound not run it.

### Your environment
* OS Version: `win10`
* ROS Distro: `Melodic`
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
[image]
    ```
Following this issue, I cannot access any information of their version.

rhaschke, 2020-02-08:
@seanyen, can you comment on this?

seanyen, 2020-02-10:
@PackChinoros from the error messages "The procedure entry could not be located", it most likely there are some binaries version mismatched in your environment. Did you upgrade from a older install to a newer one?

You can try to nuke this folder `c:\opt\ros\melodic\x64\` and run `choco upgrade ros-melodic-desktop_full -y --force  --execution-timeout=0` again. If it doesn't help, I would suggest you to follow the instructions of `Uninstall` and then do `Binary Package Installation` again from this ROS [Wiki](http://wiki.ros.org/Installation/Windows).

PackChinoros, 2020-02-11:
Thanks for getting back to me, I've already uninstalled then I tried to use ubuntu on WSL but another problem raise as OpenGL could not detect so I deleted everything and utilize VMware instead. Now, everything works well!

closed 2020-02-11 by PackChinoros