ros-visualization/rviz #1473: Segmentation fault when compiling rviz for ROS

pilino1234, 2020-02-21:
I'm trying to compile rviz from source (a long story). Next to a number of depreciation warnings for some qt5 libary functions, I get a segfault while compiling. 

```
Starting  >>> rviz                                                                                                                  
____________________________________________________________________________________________________________________________________ 
Errors     << rviz:make /home/martin/Documents/DATX02/ROS/overlay_ws/logs/rviz/build.make.000.log
[snip]
make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:95: sip/rviz_sip/Makefile] Segmentation fault (core dumped)
make[1]: *** [CMakeFiles/Makefile2:4931: src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/martin/Documents/DATX02/ROS/overlay_ws/build/rviz; catkin build --get-env rviz | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
....................................................................................................................................
Failed     << rviz:make                                    [ Exited with code 2 ]                                                   
Failed    <<< rviz                                         [ 1 minute and 23.9 seconds ]                                            
[build] Summary: 61 of 62 packages succeeded.                                                                                       
[build]   Ignored:   7 packages were skipped or are blacklisted.                                                                    
[build]   Warnings:  8 packages succeeded with warnings.                                                                            
[build]   Abandoned: None.                                                                                                          
[build]   Failed:    1 packages failed.                                                                                             
[build] Runtime: 3 minutes and 26.4 seconds total.
```

Please find the [`build.make.000.log`](https://github.com/ros-visualization/rviz/files/4236995/build.make.000.log) and [`build.make.000.log.stderr`](https://github.com/ros-visualization/rviz/files/4236999/build.make.000.log.stderr.txt) files attached.

I've googled around and found some unsatisfying solutions such as installing everything in a virtualbox and disabling 3D acceleration.

The most related line from the stderr output seems to be:
```
make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:95: sip/rviz_sip/Makefile] Segmentation fault (core dumped)
```

I looked into the build.make file mentioned in the error, but there is nothing really suspicious about it, except that it is one long line (2.4k characters) seemingly containing all packages in my $ROS_PACKAGE_PATH.

Does anybody know where I can find continue debugging this problem or if there is a solution?
[build.make.000.log](https://github.com/ros-visualization/rviz/files/4236995/build.make.000.log)

### Your environment
* OS Version: openSUSE Leap 15.1
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    * rviz: Cloned at commit https://github.com/ros-visualization/rviz/commit/6bf59755eb213afa575e219feb152c0efd8b3209
    * Qt: 5.14.1
    * OGRE: 1.9.0
    * OpenGL: 3.0
    * Mesa 18.3.2

rhaschke, 2020-02-21:
`sip` is related to the python bindings for rviz. Maybe, you have an outdated/incompatible sip compiler? As a simple measure, I suggest to disable python bindings here:
https://github.com/ros-visualization/rviz/blob/6bf59755eb213afa575e219feb152c0efd8b3209/src/CMakeLists.txt#L8
Otherwise, you should check your sip compiler.

pilino1234, 2020-02-21:
Hi!
Thanks for the fast response. 
I have the following sip command available on my computer:
```
sip      sip-2.7  sip-3.6
```

The version is 4.19.19 for all of these.

I tried commenting out line 8 and for now the compilation seems to be running longer. What will I miss if I don't have the Python bindings?

rhaschke, 2020-02-21:
You will not be able to use rviz embedded in rqt or another python program.

closed 2020-02-21 by rhaschke