ros-visualization/rviz #1474: Display string topic

felixvd, 2020-03-03:
I have a need to display Strings to visualize the system state. There is a display for Image messages, but not for String messages. It would be convenient if there was a display type like `Image` for Strings, which simply displayed the string content in its window and had formatting options in the dropdown menu.

1) Am I overlooking a function that would make displaying strings easy, except publishing a Marker? I don't think that one is an alternative in this case.

2) Would a PR for this be welcome if I implement it myself?

rhaschke, 2020-03-03:
I'm not aware of such a display, but you might want to ask on ROS anwers first.
PRs are always welcome :wink:

felixvd, 2020-03-03:
I forgot to search on ROS Answers after googling. It looks like [this package](http://wiki.ros.org/jsk_rviz_plugins) does what I need, according to [this answer](https://answers.ros.org/question/341944/display-text-in-rviz/). My bad for the noise!

closed 2020-03-03 by felixvd