ros-visualization/rviz #1477: QXcbConnection: XCB error

surabhisarda05, 2020-03-11:
I've installed ROS Kinetic and receive this error on running rviz.
user@qpul7:~$ rviz
[ INFO] [1583963209.495153019]: rviz version 1.12.17
[ INFO] [1583963209.495240895]: compiled against Qt version 5.5.1
[ INFO] [1583963209.495256345]: compiled against OGRE version 1.9.0 (Ghadamon)
QXcbConnection: XCB error: 148 (Unknown), sequence: 170, resource id: 0, major code: 140 (Unknown), minor code: 20
Segmentation fault (core dumped)

Environment: Ubuntu 16.04.06, ROS Kinetic
OpenGL version string: 3.0 Mesa 18.0.5

I've installed ROS Kinetic &  rviz multiple times now.
How can I resolve this error?

Thanks!

rhaschke, 2020-03-12:
This is a Qt error, but not an rviz error. Search the web for similar issues, for example:
https://bugreports.qt.io/browse/QTBUG-48453

Do you use the standard Qt installation on Ubuntu 16.04? Do you connect remotely, e.g. via ssh?

surabhisarda05, 2020-03-12:
Yes, using standard QT install and do not connect via SSH. It is desktop env.

rhaschke, 2020-03-13:
I have no more ideas then. Sorry. It does work on standard Ubuntu 16.04 installations as well as with the Kinetic docker container. So, the issue is somehow related to your system.

rhaschke, 2021-02-05:
Closing due to inactivity.

closed 2021-02-05 by rhaschke

130s, 2021-05-08:
Don't know if the environment is exactly the same but I'm running into this error with RViz. See my report to rocker https://github.com/osrf/rocker/issues/146 where I also reported a dumb workaround.

rhaschke, 2021-05-12:
The error code reported here (148) is different from the one reported at rocker (2). I think these issues are unrelated.

130s, 2021-05-13:
> The error code reported here (148) is different from the one reported at rocker (2). I think these issues are unrelated.

Oh I see. I only looked at the title of this ticket. If the difference of error code matters (I'm not familiar with Qt these days so no idea) I suggest to rename the title to avoid posting on a wrong thread in the future.