ros-visualization/rviz #1478: Crash while constructing rviz::shape Cube

SvenMuc, 2020-03-13:
I create multiple ''rviz::shape' instances in my own RVIZ Plugin.

```c++
    for (int i = 0; i < message_.configuration.number_sectors; ++i)
    {
      auto sector_blocked_area = new rviz::Shape(rviz::Shape::Sphere, scene_manager_, frame_node_);
      shape_blocked_sector_area_.push_back(sector_blocked_area);
    }
```
So far so good. Everything works as expected. As soon as I change from Sphere to Cube it crashed in the first `new` call with a segmentation fault.

```c++
    for (int i = 0; i < message_.configuration.number_sectors; ++i)
    {
      auto sector_blocked_area = new rviz::Shape(rviz::Shape::Cube, scene_manager_, frame_node_);
      shape_blocked_sector_area_.push_back(sector_blocked_area);
    }
```
### GDB Backtrace
https://gist.github.com/SvenMuc/551a02468768953a4b83d054b1f95603

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
   [ INFO] [1584115063.384621578]: rviz version 1.13.7
   [ INFO] [1584115063.384660411]: compiled against Qt version 5.9.5
   [ INFO] [1584115063.384670758]: compiled against OGRE version 1.9.0 (Ghadamon)
   [ INFO] [1584115063.399693503]: Forcing OpenGl version 0.

    ```
* If source build, which git commit?

rhaschke, 2020-03-13:
I have seen similar (OGRE / OpenGL) errors in a different context before. For sure, it's not related to rviz::Shape::Cube. Could you try the noetic-devel branch from [here](https://github.com/rhaschke/rviz/tree/noetic-devel) with a more recent Ogre version (> 1.12)?

SvenMuc, 2020-03-16:
> Could you try the noetic-devel branch from [here](https://github.com/rhaschke/rviz/tree/noetic-devel) with a more recent Ogre version

How do I install a separate RVIZ version? I cloned the repo you mentioned and build it. So far so good. But when I run RVIZ, I still get the old version.

rhaschke, 2020-03-16:
You just need to source your workspace.

SvenMuc, 2020-03-16:
This is what I did. I cloned your linked repo into my actual catkin workspace and build it with `catkin build rviz`. Finally I sourced my workspace and run `rviz`.
But I still get the old version numbers in the console and the same RVIZ crash as described above.

rhaschke, 2020-03-16:
You also need to build a newer Ogre version (first!) from: https://github.com/OGRECave/ogre.
To use this ogre version by your catkin workspace, specify the same _install_ dirs for your Ogre and your catkin build.

SvenMuc, 2020-03-17:
Ok, I managed to build OGRE on my machine after some while. But I still struggling to with the correct libzzip when I build rviz. Any ideas? The whole build process seems to be not well documented.

```
make[2]: *** [/home/i009658/workspace/autonomous_tractor/project/ros/devel/.private/rviz/lib/rviz/render_points_test] Error 1
make[1]: *** [src/test/CMakeFiles/render_points_test.dir/all] Error 2
/usr/bin/ld: warning: libzzip.so.13, needed by /usr/local/lib/libOgreMain.so, not found (try using -rpath or -rpath-link)
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_strerror_of'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_dirhandle'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_dir_open_ext_io'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_file_open'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_error'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_init_io'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_dir_close'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_tell'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_seek'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_file_close'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_file_read'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_dir_read'
/usr/local/lib/libOgreMain.so: undefined reference to `zzip_dir_stat'
collect2: error: ld returned 1 exit status
```

rhaschke, 2020-03-17:
I remember having issues like these as well. I think, I forced ogre to use the system library in /usr/lib/x86_64-linux-gnu/libzzip.so, using `-DOGRE_BUILD_DEPENDENCIES=OFF`. Do you have this lib installed?

SvenMuc, 2020-03-17:
This I already tried. Still same issues.

rhaschke, 2020-03-17:
Sorry, I'm afraid I cannot help then anymore.

SvenMuc, 2020-03-17:
Is there any plan to officially update rviz and OGRE?

rhaschke, 2020-03-17:
Releasing a new Ogre version into the Ubuntu system is a tedious process, which @simonschmeisser started for the upcoming Ubuntu 20.04. I'm not sure the Ubuntu maintainer team accepted this request... If so, the plan is to switch to the new Ogre version in Noetic.
Note, there is another, very much related issue: https://github.com/ros-visualization/rviz/issues/1456.

simonschmeisser, 2020-03-17:
the process for updating packages is to first get them updated in Debian and then have them pulled into Ubuntu. Since we went for a parallel installation ogre-1.12 is considered a new package and currently sits in the NEW queue for more than a month. I am told there is no way to push things.

---

More on topic, your crash happens in thread 15 which I assume is not the main gui thread. Ogre and rviz are not really thread-safe. You could try adding some blocking/unblocking before and after you add to the scene. Or you could try using Qt's mechanisms and put the function into a SLOT which you call from your callback function. Either by emitting a SIGNAL or by using QObject::invokeMessage ( queued connection). Your slot will then be called in the event loop of the main thread (ie between rendering) 
https://doc.qt.io/qt-5/signalsandslots.html

rhaschke, 2020-03-17:
@simonschmeisser, thanks for pointing out the potential threading issue. Indeed the issue occurs in the thread serving the threaded callback queue.

SvenMuc, 2020-06-29:
Thank you @simonschmeisser and @rhaschke for the detailed feedback. Finally, I solved the issue by introducing the SIGNAL/SLOT mechanism. 

Nevertheless, if you strictly follow the tutorials on [ROS/RVIZ tutorials]( https://github.com/ros-visualization/visualization_tutorials/tree/noetic-devel/rviz_plugin_tutorials) this solution is not obvious. Maybe we should adopt the tutorials as well.

simonschmeisser, 2020-06-29:
Glad that we could help you! We'll happily review PRs for the tutorials!

rhaschke, 2020-06-29:
You are welcome to file PRs to improve the tutorial(s). Hope, your issue is resolved. Closing this issue for now.

closed 2020-06-29 by rhaschke