ros-visualization/rviz #1479: rviz_assimp_vendor CMake CMP0012 build error

rotu, 2020-03-18:
CMake 3.16.3 on Ubuntu Focal, building from Master

```
--- stderr: rviz_assimp_vendor                                                            
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimpTargets.cmake:54 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimp-config.cmake:1 (include)
  CMakeLists.txt:55 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
```

rhaschke, 2020-04-06:
This is an issue with rviz for ROS2. Please, file the issue [there](https://github.com/ros2/rviz). Here is the ROS1 repository.
I even think that this error should be reported to assimp itself.

closed 2020-04-06 by rhaschke

rotu, 2020-04-06:
Oops! Thank you!