ros-visualization/rviz #1485: Origin of new sphere marker has an offset

Levaru, 2020-03-31:
I noticed that since the new commit 2cf0b247e81b82899b2616bee9008356b5ab0f62 the sphere marker has a light offset, most likely caused by a wrong origin of the mesh when it was created.

You can barely see it in this screenshot, but it is quite noticeable in my application.
[image]

### My environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO]: rviz version 1.13.7
    [ INFO]: compiled against Qt version 5.9.5
    [ INFO]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO]: Forcing OpenGl version 0.
    [ INFO]: Stereo is NOT SUPPORTED
    [ INFO]: OpenGl version: 3 (GLSL 1.3).
    ```

rhaschke, 2020-03-31:
@wxmerkt, please can you have a look at this issue as you authored the original PR #1463.

wxmerkt, 2020-03-31:
Yes, I will take a look at it.

wxmerkt, 2020-03-31:
Please take a look at #1487 - I fixed the issue with the sphere being off centre.

PR #1487 (2020-03-31): Fix sphere being off centre, smooth cube
This PR fixes #1485 (sphere being off centre). I have also added a smooth cube which was not included in the original PR.

## Original sphere (e.g. `kinetic-devel`)
[image]

## Sphere in current `melodic-devel` (which is off-centre)
[image]

## Fixed sphere in this PR
[image]

---

## Current cube
[image]

## New cube
[image]
closed 2020-03-31 by Levaru