ros-visualization/rviz #1490: RVIZ Segmentation Fault on WSL2

malghooneh, 2020-04-21:
I am running rosrun rviz rviz in WSL2 and keep getting "segmentation fault" .. 

[image]

### My environment is:
WSL2 (ubuntu 18.04) on windows 10
ROS Melodic

I have set VcXSrv on the windows side, and glxgears works fine ...

Any idea what else I can try?

rhaschke, 2020-04-26:
See https://github.com/ros-visualization/rviz/issues/1491#issuecomment-619532542 for a possible answer. What is the difference between WSL ans WSL2?

malghooneh, 2020-04-30:
> See [#1491 (comment)](https://github.com/ros-visualization/rviz/issues/1491#issuecomment-619532542) for a possible answer. What is the difference between WSL ans WSL2?

Thanks for the link. The main difference is Kernel. WSL 2 uses an entirely new architecture that uses a real Linux kernel.

closed 2020-05-08 by rhaschke