ros-visualization/rviz #1491: rviz segfaulting - Windows 10 + Ubuntu 16.04 + Kinetic

engnfrc, 2020-04-24:
Dell Latitude E6420
OS Version: Microsoft Windows 10 Enterprise + Windows Subsystem for Linux using Ubuntu 16.04
ROS Distro: Kinetic (Installed from Packages on Apt)

Dual Monitor setup on a laptop with integrated and dedicated graphics:
Intel(R) HD Graphics 3000 - Driver Intel 5/19/2016 9.17.10.4459
NVIDIA NVS 4200M - Driver NVIDIA 7/22/2015 10.18.13.5362

rviz -h
[ INFO] [1587746430.188146600]: rviz version 1.12.17
[ INFO] [1587746430.188677700]: compiled against Qt version 5.5.1
[ INFO] [1587746430.188866300]: compiled against OGRE version 1.9.0 (Ghadamon)

glxinfo | grep "OpenGL version"
OpenGL version string: 1.2 (4.5.0 NVIDIA 353.62)

rviz is now segfaulting when I try to use it. It's worked fine all week up since I installed all of this in Windows until after the restart I did this morning. It still is able to run if I "sudo su" first. I'm using Xming to get the display up. I export DISPLAY=:0 before starting rviz. Updating display drivers from device manager says the driver is up to date. Before I went to restart there was no mention of "Update" in the start menu. I did try "export OGRE_RTT_MODE=Copy" before running and the other 2 options; none of the 3 affected the behavior. When trying to run: "rosrun rviz system_info.sh", system_info.sh wasn't found, nor is it in /opt/ros/kinetic. No core file is being generated. Other requested logs are attached. Any ideas? Maybe a temp file relating to my user that's corrupt and needs flushing?

[rviz.log](https://github.com/ros-visualization/rviz/files/4530207/rviz.log)
[rviz-gdb.log](https://github.com/ros-visualization/rviz/files/4530208/rviz-gdb.log)
[rviz-l.log](https://github.com/ros-visualization/rviz/files/4530209/rviz-l.log)

rhaschke, 2020-04-26:
As mentioned [here](https://janbernloehr.de/2017/06/10/ros-windows), in `VcXsrv `you need to disable the default setting `Native opengl`. Following this instruction, rviz works fine for me.
Probably, a corresponding setting is required for `Xming` as well.

rhaschke, 2020-04-26:
Duplicate of #1490.

closed 2020-04-26 by rhaschke