ros-visualization/rviz #1498: Noetic release

rhaschke, 2020-04-27:
This issue shall track open issues for the upcoming Noetic release.

- [x] Release upstream packages
- [x] Remove support for really old upstream dependencies (and corresponding `#ifdefs`)
  - [x] Ogre < 1.9 (811345857e380da3a6f624b75019e28d1e7f63a5)
  - [x] assimp < 2.0 (bc8214d4b3af1ebea709082353f153546c4676c5)
  - [x] boost::filesystem < 3 (0abad52b546d6afedd31ab8786f629df003cd60d)
- [ ] Finish tf2 migration (#1376)
  I started this work in #1497, but cannot continue for the rest of the week. Any help is appreciated.
  @sloretz: We could also skip this for now and have another (ABI and API-breaking) release in 1-2 weeks. If agreed, I could prepare this preliminary release tomorrow.
- [ ] Cleanup for Ogre 1.12
  Thanks to @simonschmeisser Ubuntu 20.04 ships with Ogre 1.12 - additionally to old Ogre 1.9. However, as discussed in https://github.com/ros/rosdistro/pull/24448, Gazebo still relies on Ogre 1.9. We could build RViz against Ogre 1.12, but this would require people to switch versions of `libogre-dev` between 1.9 and 1.12 when switching between the development of Gazebo and RViz.
My suggestion is to postpone this step and wait for the Gazebo team (@chapulina) to migrate as well. The current code base is compatible with all Ogre versions between 1.9 and 1.12.
- [x] clang-tidy (#1494, 71715ceacf0b8344873659e339af23cf1fb2aa1e)

rhaschke, 2020-05-03:
I think I finished the preparation for Noetic. My "release candidate" is here:
https://github.com/rhaschke/rviz/tree/noetic-devel
@simonschmeisser, please test.

simonschmeisser, 2020-05-03:
```
In file included from src/rviz/ogre_helpers/movable_text.cpp:42:0:
src/rviz/ogre_helpers/movable_text.h:204:23: error: ‘const String& rviz::MovableText::getName() const’ marked ‘override’, but does not override
   const Ogre::String& getName() const override
                       ^~~~~~~

```
using ogre 1.12.4

wjwwood, 2020-05-06:
@rhaschke let me know if I can help with this, I have a bit of time for Noetic work this week. <3

I'll start by testing your noetic branch.

rhaschke, 2020-05-06:
Thanks for offering your help, @wjwwood. I have run all the test binaries and scripts of rviz, but another, independent tester would be great. If you give it a go, I will prepare the release later on.

rhaschke, 2020-05-06:
@simonschmeisser, the Ogre 1.12 build issue was fixed via #1503.

wjwwood, 2020-05-06:
Sure. I checked the basics. Only thing so far is that we will need to setup the rosdep rules for ogre and ogre dev.

simonschmeisser, 2020-05-06:
I can confirm the basics are working as well on ogre-1.12, make run_tests also succeeds. Where can I find the instructions for running the render tests?

rhaschke, 2020-05-06:
There are no explicit instructions. I just run all the binaries/scripts that are available via `rosrun rviz ...`
To validate, one needs to configure an appropriate display plugin...

simonschmeisser, 2020-05-06:
Hmm, inconvenient ...

```
rosrun rviz color_editor_test 
QObject::connect: Cannot connect QColorDialog::currentColorChanged(const QColor&) to (nullptr)::update()
double free or corruption (out)
```
after selecting a color ... but in rviz proper it works just fine

wjwwood, 2020-05-06:
Re: Cleanup for Ogre 1.12

I commented on the ros/rosdistro pr as well (https://github.com/ros/rosdistro/pull/24448#issuecomment-624929858), but I think we cannot actually do 1.12 so long as gazebo depends on 1.9, because if we did then rviz and gazebo_ros_pkgs would not be able to be installed at the same time, as one would depend on `libogre-1.9-dev` and the other would depend on `libogre-1.12-dev` and those are not able to be installed simultaneously:

```
william@ubuntu:~/rviz_ws$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal
william@ubuntu:~/rviz_ws$ sudo apt install libogre-1.9-dev libogre-1.12-dev
[sudo] password for william:
Reading package lists... Done
Building dependency tree
Reading state information... Done
libogre-1.9-dev is already the newest version (1.9.0+dfsg1-12build1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libogre-1.12-dev : Conflicts: libogre-1.9-dev but 1.9.0+dfsg1-12build1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

wjwwood, 2020-05-06:
Also, I think there are some issues with some of the tutorials, for example I was spot checking the python tutorial and it isn't working at the moment (even after fixing up for python3):

```
william@ubuntu:~/rviz_ws$ ./install_isolated/lib/rviz_python_tutorial/myviz.py 
Traceback (most recent call last):
  File "./install_isolated/lib/rviz_python_tutorial/myviz.py", line 161, in <module>
    myviz = MyViz()
  File "./install_isolated/lib/rviz_python_tutorial/myviz.py", line 46, in __init__
    self.frame = rviz.VisualizationFrame()
AttributeError: module 'rviz' has no attribute 'VisualizationFrame'
```

So, I'm investigating that as I have time too, but it also looks like the test scripts in this repository still do `#!/usr/bin/env python`, which will need to be changed in order to properly use python3, I believe:

```diff
william@ubuntu:~/rviz_ws/src/visualization_tutorials$ git diff
diff --git a/rviz_python_tutorial/myviz.py b/rviz_python_tutorial/myviz.py
index 62b7203..80067e1 100755
--- a/rviz_python_tutorial/myviz.py
+++ b/rviz_python_tutorial/myviz.py
@@ -1,4 +1,4 @@
-#!/usr/bin/env python
+#!/usr/bin/env python3
 
 ## BEGIN_TUTORIAL
 ##
```

Sorry if there are already issues made for these items, I haven't had a chance to read the issue backlog yet.

rhaschke, 2020-05-07:
This is due to a small API change in the python wrapper here: f3eeaf4c695ed6bb1ef990e3cc37fb23801e5c1b.

Adapting the import line from
`import rviz`
to
`from rviz import bindings as rviz`
will fix the problem. We need to announce this API change somewhere though...

sloretz, 2020-05-07:
> We need to announce this API change somewhere though...

@rhaschke [added to the Noetic migration guide](http://wiki.ros.org/noetic/Migration#RViz_Python_API_import). Look correct?

rhaschke, 2020-05-08:
Great, thanks @sloretz!

wjwwood, 2020-05-11:
I'm going to continue testing this afternoon with ogre-1.9, since I made the decision to stick with that for Noetic.

@rhaschke I don't want to step on your feet, but I would like to get rviz released sooner than later. How can I help other than testing? Do you need to me to do any of the release bits? (changelogs, tagging, bloom, etc...)

rhaschke, 2020-05-11:
Dear @wjwwood, I had already prepared most of these release bits locally. I was just waiting for the decision on Ogre 1.9 vs. 1.12. I'm currently running bloom already.
I did extensive testing a week ago and everything looked good to me. If you find issues in your tests, I will be happy to file a fixup release :wink:
This is the release PR: https://github.com/ros/rosdistro/pull/24810

wjwwood, 2020-05-11:
Amazing, thanks so much @rhaschke!

closed 2020-05-12 by wjwwood