ros-visualization/rviz #1499: Export From Rviz

Juuustin, 2020-05-01:
Hi, Thank you for your excellent work.
I am running a SLAM program using Rviz based on ROS to show the points cloud. I want to save the showed map in Rviz, is there a method to save the points showed directly?
Looking forward to hearing from you soon!

rhaschke, 2020-05-01:
No, there is no saving option. rviz is a real-time visualizer. Which point in time should be saved?
If you want to save your data consider [rosbag](http://wiki.ros.org/rosbag).

closed 2020-05-01 by rhaschke

Juuustin, 2020-05-01:
> No, there is no saving option. rviz is a real-time visualizer. Which point in time should be saved?
> If you want to save your data consider [rosbag](http://wiki.ros.org/rosbag).

I'd like to save the final output of the result, that is, the showed points after running, could you recommend a rosbag method for it? Thank you very much!

rhaschke, 2020-05-01:
Sorry, not sure what you are referring to. Are you talking about the rviz display accumulating points that are sent incrementally by your PCL publisher and you want to save the final result, i.e. all of them? I don't know of an option to do this. First question would be, which file format do you expect?
I suggest writing a simple python script accumulating the data and writing them to your preferred file format.

Juuustin, 2020-05-05:
Sorry for the unclear question. Because I use '--img_fov' so there is only a few points in front of ego car, I want to view the whole lidar points, now I just delete the '--img_fov', it seems that the whole 360-degree view is presented.