ros-visualization/rviz #1508: noetic rviz cannot change point size for pointcloud2 topic

flynneva, 2020-05-26:
looks like in the noetic rviz release there seems to be a bug when you try to change the point size for pointcloud2 topics.

Here is a screenshot of the default pointcloud points that show up when I add in the viz, and yes I have tried to change the type/size of it:
[image]

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
[ INFO] [1590534094.203451371]: rviz version 1.14.0
[ INFO] [1590534094.203502146]: compiled against Qt version 5.12.8
[ INFO] [1590534094.203514196]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1590534094.212776806]: Forcing OpenGl version 0.
[ INFO] [1590534094.416371615]: Stereo is NOT SUPPORTED
[ INFO] [1590534094.416444253]: OpenGl version: 4.6 (GLSL 4.6).
    ```

rhaschke, 2020-05-27:
Sorry, but I cannot reproduce your issue. Showing the test point cloud generated by `rosrun rviz send_point_cloud_2`, I get the expected results and point size can be changed:

[image]

doisyg, 2020-05-27:
A bit similar here, points from pcl or LaserScan appears very small, size cannot be changed and with no color, unless rviz is started with `export LIBGL_ALWAYS_SOFTWARE=1`
Ubuntu 20.04 and Intel x86 computer with no discrete GPU, maybe openGL or gpu driver related

rhaschke, 2020-05-27:
This is weird. I tested on two different laptops, no discrete GPU as well. No problems.

flynneva, 2020-05-28:
I thought it was weird that the console output printed "Forcing OpenGL version 0". is that normal output?

doisyg, 2020-05-28:
On my laptop with a discrete nvidia GPU, no issue but on another computer Intel only, same issue (all points are small and colorless)

flynneva, 2020-05-28:
my computer has an Intel as well.

rhaschke, 2020-05-29:
I'm running on an Intel "GPU" as well. However, the OpenGL version reported is different for me:
```
[ INFO] ros.rviz: rviz version 1.14.0
[ INFO] ros.rviz: compiled against Qt version 5.12.8
[ INFO] ros.rviz: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] ros.rviz: Forcing OpenGl version 0.
[ INFO] ros.rviz: Stereo is NOT SUPPORTED
[ INFO] ros.rviz: OpenGl version: 3 (GLSL 1.3).
```
Could you please report your installed (software) OpenGL packages? Mine are:
```bash
$ dpkg -l | grep mesa
ii  libegl-mesa0:amd64                        20.0.4-2ubuntu1                     amd64        free implementation of the EGL API -- Mesa vendor library
ii  libgl1-mesa-dev:amd64                     20.0.4-2ubuntu1                     amd64        transitional dummy package
ii  libgl1-mesa-dri:amd64                     20.0.4-2ubuntu1                     amd64        free implementation of the OpenGL API -- DRI modules
ii  libglapi-mesa:amd64                       20.0.4-2ubuntu1                     amd64        free implementation of the GL API -- shared library
ii  libglu1-mesa:amd64                        9.0.1-1build1                       amd64        Mesa OpenGL utility library (GLU)
ii  libglu1-mesa-dev:amd64                    9.0.1-1build1                       amd64        Mesa OpenGL utility library -- development files
ii  libglx-mesa0:amd64                        20.0.4-2ubuntu1                     amd64        free implementation of the OpenGL API -- GLX vendor library
ii  mesa-utils                                8.4.0-1build1                       amd64        Miscellaneous Mesa GL utilities
ii  mesa-va-drivers:amd64                     20.0.4-2ubuntu1                     amd64        Mesa VA-API video acceleration drivers
ii  mesa-vdpau-drivers:amd64                  20.0.4-2ubuntu1                     amd64        Mesa VDPAU video acceleration drivers
ii  mesa-vulkan-drivers:amd64                 20.0.4-2ubuntu1                     amd64        Mesa Vulkan graphics drivers

```

doisyg, 2020-05-29:
On the Intel platform:

> $ rviz 
> [ INFO] [${node}]: rviz version 1.14.0
> [ INFO] [${node}]: compiled against Qt version 5.12.8
> [ INFO] [${node}]: compiled against OGRE version 1.9.0 (Ghadamon)
> [ INFO] [/rviz_1590744687884621354]: Forcing OpenGl version 0.
> [ INFO] [/rviz_1590744687884621354]: Stereo is NOT SUPPORTED
> [ INFO] [/rviz_1590744687884621354]: OpenGl version: 4.6 (GLSL 4.6).
> 

and
```

$ dpkg -l | grep mesa
ii  libegl-mesa0:amd64                          20.0.4-2ubuntu1                       amd64        free implementation of the EGL API -- Mesa vendor library
ii  libegl1-mesa-dev:amd64                      20.0.4-2ubuntu1                       amd64        free implementation of the EGL API -- development files
ii  libgl1-mesa-dev:amd64                       20.0.4-2ubuntu1                       amd64        transitional dummy package
ii  libgl1-mesa-dri:amd64                       20.0.4-2ubuntu1                       amd64        free implementation of the OpenGL API -- DRI modules
ii  libglapi-mesa:amd64                         20.0.4-2ubuntu1                       amd64        free implementation of the GL API -- shared library
ii  libglu1-mesa:amd64                          9.0.1-1build1                         amd64        Mesa OpenGL utility library (GLU)
ii  libglu1-mesa-dev:amd64                      9.0.1-1build1                         amd64        Mesa OpenGL utility library -- development files
ii  libglx-mesa0:amd64                          20.0.4-2ubuntu1                       amd64        free implementation of the OpenGL API -- GLX vendor library
ii  mesa-va-drivers:amd64                       20.0.4-2ubuntu1                       amd64        Mesa VA-API video acceleration drivers
ii  mesa-vdpau-drivers:amd64                    20.0.4-2ubuntu1                       amd64        Mesa VDPAU video acceleration drivers
ii  mesa-vulkan-drivers:amd64                   20.0.4-2ubuntu1                       amd64        Mesa Vulkan graphics drivers

```

doisyg, 2020-05-29:
```
$ clinfo
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.1 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_fp64 cl_khr_subgroups cl_khr_il_program cl_intel_spirv_device_side_avc_motion_estimation cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory_preview cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_intel_motion_estimation cl_intel_device_side_avc_motion_estimation cl_intel_advanced_motion_estimation cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_intel_media_block_io cl_khr_3d_image_writes cl_intel_va_api_media_sharing 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel(R) OpenCL HD Graphics
Number of devices                                 1
  Device Name                                     Intel(R) Gen9 HD Graphics NEO
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.1 NEO 
  Driver Version                                  20.20.16837
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               24
  Max clock frequency                             1150MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None, None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              32
  Max sub-groups per work group                   32
  Sub-group sizes (Intel)                         8, 16, 32
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 1 / 1       
    half                                                 8 / 8        (cl_khr_fp16)
    float                                                1 / 1       
    double                                               1 / 1        (cl_khr_fp64)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Single-precision Floating-point support         (core)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  Yes
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Address bits                                    64, Little-Endian
[... 65 lines cut ...]

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Intel(R) OpenCL HD Graphics
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [INTEL]
  clCreateContext(NULL, ...) [default]            Success [INTEL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
[... cut ...]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Intel(R) OpenCL HD Graphics
    Device Name                                   Intel(R) Gen9 HD Graphics NEO

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1

```

doisyg, 2020-05-29:
Interesting, we starting with with `$ rviz --opengl 210` , it works !

doisyg, 2020-05-29:
Last bunch of info:
```
$ glxinfo|grep OpenGL
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) UHD Graphics (CML GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.0.4
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.0.4
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.0.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```
So it looks like a OpenCL version issue

rhaschke, 2020-05-29:
I'm puzzled why I get OpenGL 3 on my machine, while you get OpenGL 4.6 on yours - having the same mesa libs installed!

doisyg, 2020-05-29:
To add to the confusion, on my nvidia machine, rviz works fine and it seems that I have openGL 4.6:
```
glxinfo|grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1070/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 440.64
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 440.64
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 440.64
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

```

About the different version between mine and your Intel machine, it may be because we have a different generation and also a different driver. I use the Intel NEO: https://github.com/intel/compute-runtime/releases

rhaschke, 2020-05-29:
That's not so surprising to me. I think we need to blame the 4.6 implementation of libmesa...
nvidia's OpenGL implementation might just work fine.

rhaschke, 2020-05-29:
@wjwwood, can you delegate this issue to an OpenGL expert? I neither have the experience nor the resources to handle this.

wjwwood, 2020-05-29:
I can ask @iche033 if he's got any ideas, but I don't know any opengl experts other than him (I'm certainly not one, haha). The other avenue would be to produce an example that using Ogre only and ask them about it. I'll see if I can allocate some time to help with that, but it will be a few weeks most likely as we're in the midst of our second release in so many weeks, and there will be fallout to contend with after.

wjwwood, 2020-05-29:
I do know that in my virtual machines I only have OpenGL 2.something, and that's not enough for Ogre any longer (it used to be) and so it throws when it cannot get a certain OpenGL function it needs. I don't think it's the same issue, but it is not the first time we've had issues with different opengl versions.

wjwwood, 2020-05-29:
Also, it might be worth trying to build rviz from source using the newer ogre version, as it might already be fixed by their abstraction layer?

rhaschke, 2020-05-29:
> Also, it might be worth trying to build rviz from source against the newer ogre version 1.12.

That's actually a good idea. @doisyg or @flynneva do you have resources to do so and report the result?

doisyg, 2020-05-29:
I have this error when compiling on Focal from src and I believe it is ogre version 1.9 which is used:
```
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for modules 'OGRE;OGRE-Overlay'
--   Found OGRE, version 1.9.0
--   Found OGRE-Overlay, version 1.9.0
-- OGRE_LIBRARIES=/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;/usr/lib/x86_64-linux-gnu/libOgreMain.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- OGRE_INCLUDE_DIRS=/usr/include/OGRE/Overlay;/usr/include/OGRE
-- OGRE_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/OGRE-1.9.0
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so

```

wjwwood, 2020-05-29:
Yeah, we decided to stick with 1.9 for our debs, but 1.12 is also available via `libogre-dev-1.12` (I think) and rviz can be compiled against that too.

So you'd need to uninstall `libogre-dev-1.9` and install `libogre-dev-1.12` and then build rviz from scratch (delete existing build folder or ensure CMake is rerun).

doisyg, 2020-05-29:
After compilation with ogre 1.12:

> $rosrun rviz rviz 
> [ INFO] [1590780380.302481058]: rviz version 1.14.0
> [ INFO] [1590780380.302609856]: compiled against Qt version 5.12.8
> [ INFO] [1590780380.302650942]: compiled against OGRE version 1.12.4 (Rhagorthua)
> [ INFO] [1590780380.325357004]: Forcing OpenGl version 0.
> [ INFO] [1590780380.626125246]: Stereo is NOT SUPPORTED
> [ INFO] [1590780380.626204835]: OpenGl version: 4.6 (GLSL 4.6).
> [ERROR] [1590780380.639836601]: Error: ScriptCompiler - unexpected token in liberation_sans.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
> 
Rviz starts, then crash when trying to display a pointcloud:

> [ERROR] [1590780429.109706877]: No techniques available for material [PointCloudMaterial0Sphere]
> [ERROR] [1590780429.110194420]: No techniques available for material [PointCloudMaterial0FlatSquare]
> [ERROR] [1590780429.619428720]: No techniques available for material [PointCloudMaterial1Sphere]
> [ERROR] [1590780429.619645848]: No techniques available for material [PointCloudMaterial1FlatSquare]
> [ERROR] [1590780430.098322011]: No techniques available for material [PointCloudMaterial2Sphere]
> [ERROR] [1590780430.098555597]: No techniques available for material [PointCloudMaterial2FlatSquare]
> [ERROR] [1590780430.608390163]: No techniques available for material [PointCloudMaterial3Sphere]
> [ERROR] [1590780430.608440268]: No techniques available for material [PointCloudMaterial3FlatSquare]
> [ERROR] [1590780431.089843225]: No techniques available for material [PointCloudMaterial4Sphere]
> [ERROR] [1590780431.089972989]: No techniques available for material [PointCloudMaterial4FlatSquare]
> [ERROR] [1590780431.601881442]: No techniques available for material [PointCloudMaterial5Sphere]
> [ERROR] [1590780431.602124884]: No techniques available for material [PointCloudMaterial5FlatSquare]
> [ERROR] [1590780432.113293590]: No techniques available for material [PointCloudMaterial6Sphere]
> [ERROR] [1590780432.113444678]: No techniques available for material [PointCloudMaterial6FlatSquare]
> [ERROR] [1590780432.593522391]: No techniques available for material [PointCloudMaterial7Sphere]
> [ERROR] [1590780432.593660312]: No techniques available for material [PointCloudMaterial7FlatSquare]
> [ERROR] [1590780433.106866094]: No techniques available for material [PointCloudMaterial8Sphere]
> [ERROR] [1590780433.107107751]: No techniques available for material [PointCloudMaterial8FlatSquare]
> [ERROR] [1590780433.599885477]: No techniques available for material [PointCloudMaterial9Sphere]
> [ERROR] [1590780433.599973180]: No techniques available for material [PointCloudMaterial9FlatSquare]
> [ERROR] [1590780434.097377403]: No techniques available for material [PointCloudMaterial10Sphere]
> [ERROR] [1590780434.097551561]: No techniques available for material [PointCloudMaterial10FlatSquare]
> Segmentation fault (core dumped)
>

rhaschke, 2020-05-30:
Disappointing. I won't have time to look into this next week. 
@simonschmeisser, could you have a look? You are working with Ogre 1.12 more actively I guess.

simonschmeisser, 2020-05-30:
```
glxinfo | grep OpenGL
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 620 (Kaby Lake GT2) 
OpenGL core profile version string: 4.5 (Core Profile) Mesa 19.2.8
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 19.2.8
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 19.2.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```

I don't have a ubuntu 20.04 installed yet but on 18.04 with ogre 1.12.4 and noetic-devel it works

Main difference between the glxinfo's you posted is that @rhaschke 's notebook from 2010 and my notebook from 2016 report OpenGL 3 compatibility profile while the one from @doisyg from 2018 talks only about OpenGL 4.6. So maybe someone found it clever to drop some compatibility mode (or maybe it needs to be requested differently/manually ...) Apple apparently does not support compatibility modes at all, I thought you're using that @wjwwood ?

Seeing that --opengl 210 works, maybe we should default to that if something newer is found?

ogre has a backend/`RenderSystem` for OpenGL newer than the old-school OpenGL3 backend  which is called `GL3Plus` but using it requires some modifications to the shaders in the material folder. I failed doing those modifications ... but I can open a WIP PR that enables the backend and then maybe someone who knows anything about shaders at all can continue on that (maybe we could also ask on discourse ...)

There's also a high-level shader system that would allow using all of the backends (Metal, DX11, ...) but I haven't understood how that works yet and if it can actually handle geometry shaders (and I'm not sure to what extend that's available in 2013's ogre 1.9)

rhaschke, 2020-05-30:
Given the fact that NVIDIA's OpenGL 4.6 renders correctly, while Mesa's not, I'm still tending to blame Mesa. @doisyg, could you have a look for issue reports over there?

flynneva, 2020-05-30:
gave it a go with ogre 1.12....

```
rosrun rviz rviz
[ INFO] [1590852307.372457308]: rviz version 1.14.0
[ INFO] [1590852307.372531006]: compiled against Qt version 5.12.8
[ INFO] [1590852307.372545986]: compiled against OGRE version 1.12.4 (Rhagorthua)
[ INFO] [1590852307.388516299]: Forcing OpenGl version 0.
[ INFO] [1590852307.595392886]: Stereo is NOT SUPPORTED
[ INFO] [1590852307.595582330]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1590852307.607582822]: Error: ScriptCompiler - unexpected token in liberation_sans.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
[ERROR] [1590852326.258043428]: No techniques available for material [PointCloudMaterial0Sphere]
[ERROR] [1590852326.258421743]: No techniques available for material [PointCloudMaterial0FlatSquare]
[ERROR] [1590852327.280323290]: No techniques available for material [PointCloudMaterial1Sphere]
[ERROR] [1590852327.280439542]: No techniques available for material [PointCloudMaterial1FlatSquare]
[ERROR] [1590852328.338720758]: No techniques available for material [PointCloudMaterial2Sphere]
[ERROR] [1590852328.338899048]: No techniques available for material [PointCloudMaterial2FlatSquare]
[ERROR] [1590852329.391346141]: No techniques available for material [PointCloudMaterial3Sphere]
[ERROR] [1590852329.391653767]: No techniques available for material [PointCloudMaterial3FlatSquare]
```

flynneva, 2020-05-30:
strange...just relaunched rviz and I can now resize the points?? still getting the weird "no techniques available for material" error

rhaschke, 2020-05-30:
@flynneva, did you also get a segfault the first time you started rviz?
The material techniques might be an unrelated issue. Need to check this, when I will have some time... 
The ScriptCompiler error is known. Consider it a warning only. This is due to a different syntax in the font specification between Ogre 1.9 and Ogre 1.12. To maintain compatibility, we kept both definitions and Ogre 1.12 is complaining about the older one...

rhaschke, 2020-05-30:
Does it work to manually resize the rviz window?

flynneva, 2020-05-31:
@rhaschke randomly there is a segfault and rviz crashes, but most of the time it runs ok other than the material error shown above. and yes, manually resizing works great, its just the auto-sizing when I drag it to the right or to full screen when I am unable to resize the points.

rhaschke, 2020-05-31:
Sounds like an access to uninitialized memory or stack corruption or the like.
If you get crashes, it would be interesting to see the stack traces (Is it always the same? Where does it crash?) Could you build rviz with debug info and [sanitization](https://github.com/google/sanitizers/wiki/AddressSanitizer) please and report back?

cmake args: `-DCMAKE_BUILD_TYPE=RelWithDebInfo -fsanitize=address -fno-omit-frame-pointer`

doisyg, 2020-05-31:
Mesa 20.0.4 may be affected by a related bug (since 20.0.5 claims to fix it: https://www.mesa3d.org/relnotes/20.0.5.html):

>  Mesa 20.0.5 is a bug fix release which fixes bugs found since the 20.0.4 release.
> 
> Mesa 20.0.5 implements the OpenGL 4.6 API, but the version reported by glGetString(GL_VERSION) or glGetIntegerv(GL_MAJOR_VERSION) / glGetIntegerv(GL_MINOR_VERSION) depends on the particular driver being used. Some drivers don't support all the features required in OpenGL 4.6. OpenGL 4.6 is only available if requested at context creation. Compatibility contexts may report a lower version depending on each driver. 

Though I downloaded and compiled from source the 20.1 version and still have the issue. 

> Seeing that --opengl 210 works, maybe we should default to that if something newer is found?

I will be happy to do PR in this sense, maybe if opengl version is above 3.1 AND Ogre::RenderSystemCapabilities::getVendor() is Intel, THEN fall back to 2.1 ? @rhaschke and @simonschmeisser , what do you think ?

flynneva, 2020-06-01:
when running catkin_make with the args you wanted me to there seems to be some error with regards to Vector3 in OGRE.

**catkin_make with args:**
`./src/catkin/bin/catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS="-fsanitize=address -fno-omit-frame-pointer"`

```bash
In file included from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/moc_plant_flag_tool.cpp:9,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/mocs_compilation.cpp:4:
/home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/../../../../../src/visualization_tutorials/rviz_plugin_tutorials/src/plant_flag_tool.h:37:7: error: using typedef-name ‘Ogre::Vector3’ after ‘class’
   37 | class Vector3;
      |       ^~~~~~~
In file included from /usr/include/OGRE/OgreVector.h:32,
                 from /usr/include/OGRE/OgreVector3.h:1,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/frame_manager.h:41,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/display_context.h:37,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/message_filter_display.h:37,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/../../../../../src/visualization_tutorials/rviz_plugin_tutorials/src/imu_display.h:36,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/UVLADIE3JM/moc_imu_display.cpp:9,
                 from /home/flynn/dev/ros/noetic/build/visualization_tutorials/rviz_plugin_tutorials/rviz_plugin_tutorials_autogen/mocs_compilation.cpp:3:
/usr/include/OGRE/OgrePrerequisites.h:253:29: note: ‘Ogre::Vector3’ has a previous declaration here
  253 |     typedef Vector<3, Real> Vector3;
      |                             ^~~~~~~
```

flynneva, 2020-06-01:
tried to compile with this command according to the arguments to test above: 
`./src/catkin/bin/catkin_make --only-pkg-with-deps rviz -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS="-fsanitize=address -fno-omit-frame-pointer"
`

when I go and launch rviz....I get this & rviz crashes **only** when I **move** my mouse through the 3D view within rviz. moving the mouse and interacting with the rest of rviz works great.

<details><summary>rosrun rviz rviz</summary>
<p>

```
[ INFO] [1590975269.466012303]: rviz version 1.14.0
[ INFO] [1590975269.466104596]: compiled against Qt version 5.12.8
[ INFO] [1590975269.466131317]: compiled against OGRE version 1.12.4 (Rhagorthua)
[ INFO] [1590975269.481843791]: Forcing OpenGl version 0.
[ INFO] [1590975269.750340397]: Stereo is NOT SUPPORTED
[ INFO] [1590975269.750467646]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1590975269.783444875]: Error: ScriptCompiler - unexpected token in liberation_sans.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
=================================================================
==263546==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020002395d0 at pc 0x7fd1ac66dfdd bp 0x7ffef5da8e00 sp 0x7ffef5da8df0
READ of size 4 at 0x6020002395d0 thread T0
    #0 0x7fd1ac66dfdc in rviz::SelectionManager::unpackColors(Ogre::PixelBox const&, std::vector<unsigned int, std::allocator<unsigned int> >&) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/selection/selection_manager.cpp:593
    #1 0x7fd1ac67cec0 in rviz::SelectionManager::renderAndUnpack(Ogre::Viewport*, unsigned int, int, int, int, int, std::vector<unsigned int, std::allocator<unsigned int> >&) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/selection/selection_manager.cpp:621
    #2 0x7fd1ac67dfba in rviz::SelectionManager::pick(Ogre::Viewport*, int, int, int, int, boost::unordered::unordered_map<unsigned int, rviz::Picked, boost::hash<unsigned int>, std::equal_to<unsigned int>, std::allocator<std::pair<unsigned int const, rviz::Picked> > >&, bool) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/selection/selection_manager.cpp:894
    #3 0x7fd18bb153d6 in rviz::InteractionTool::updateFocus(rviz::ViewportMouseEvent const&) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/default_plugin/tools/interaction_tool.cpp:85
    #4 0x7fd18bb175cd in rviz::InteractionTool::processMouseEvent(rviz::ViewportMouseEvent&) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/default_plugin/tools/interaction_tool.cpp:154
    #5 0x7fd1ac7572ef in rviz::VisualizationManager::handleMouseEvent(rviz::ViewportMouseEvent const&) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/visualization_manager.cpp:539
    #6 0x7fd1ac6165df in rviz::RenderPanel::onRenderWindowMouseEvents(QMouseEvent*) /home/flynn/dev/ros/noetic/src/rviz/src/rviz/render_panel.cpp:119
    #7 0x7fd1abd432b5 in QWidget::event(QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ad2b5)
    #8 0x7fd1abd00a65 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16aa65)
    #9 0x7fd1abd0a342 in QApplication::notify(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x174342)
    #10 0x7fd1ab8d3939 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x286939)
    #11 0x7fd1abd09456 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x173456)
    #12 0x7fd1abd5f35c  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1c935c)
    #13 0x7fd1abd621eb  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1cc1eb)
    #14 0x7fd1abd00a65 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16aa65)
    #15 0x7fd1abd0a0ef in QApplication::notify(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1740ef)
    #16 0x7fd1ab8d3939 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x286939)
    #17 0x7fd1a8cbe7d2 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x1267d2)
    #18 0x7fd1a8cc010a in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x12810a)
    #19 0x7fd1a8c9a35a in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x10235a)
    #20 0x7fd1a00b932d  (/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5+0x7932d)
    #21 0x7fd1a8524fbc in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x51fbc)
    #22 0x7fd1a852523f  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x5223f)
    #23 0x7fd1a85252e2 in g_main_context_iteration (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x522e2)
    #24 0x7fd1ab92b564 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2de564)
    #25 0x7fd1ab8d24da in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2854da)
    #26 0x7fd1ab8da245 in QCoreApplication::exec() (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x28d245)
    #27 0x55679bbd099e in main /home/flynn/dev/ros/noetic/src/rviz/src/rviz/main.cpp:42
    #28 0x7fd1ab2840b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==263546==ABORTING
```
</p>
</details

rhaschke, 2020-06-01:
Thanks, @flynneva for your detailed testing and reporting. Your rviz build seems to crash due to an rviz-internal issue. Summarizing, I see several issues reported here:

1. heap overflow in rviz' SelectionManager as reported in https://github.com/ros-visualization/rviz/issues/1508#issuecomment-636570064
1. invalid materials in Ogre 1.12 as reported in https://github.com/ros-visualization/rviz/issues/1508#issuecomment-636346422
1. cannot change point size of PointCloud2 display, probably related to Mesa OpenGL 4.6
Works with Mesa OpenGL 3 and NVIDIA OpenGL 4.6. Fixed with Ogre 1.12? 
Verify again if 1. and 2. are fixed.

rhaschke, 2020-06-12:
Dear @flynneva,
I fixed the SelectionManager issue (1). Please cherry-pick #1519 to your Noetic branch and retry building with `-DCMAKE_CXX_FLAGS="-fsanitize=address -fno-omit-frame-pointer"` to verify that there are no other issues.
I was _not_ able to reproduce issues (2) and (4) summarized in https://github.com/ros-visualization/rviz/issues/1508#issuecomment-636907358. Could you please try again with the proposed fix?

flynneva, 2020-06-13:
first tried with oger 1.9.

<details><summary>rosrun rviz rviz</summary>
<p>

```
[ INFO] [1592017221.765288963]: rviz version 1.14.0
[ INFO] [1592017221.765444456]: compiled against Qt version 5.12.8
[ INFO] [1592017221.765521110]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1592017221.792844902]: Forcing OpenGl version 0.
[ INFO] [1592017222.034887503]: Stereo is NOT SUPPORTED
[ INFO] [1592017222.035347777]: OpenGl version: 4.6 (GLSL 4.6).
^C
=================================================================
==103911==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 16524 byte(s) in 33 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a6776cccc in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag) (/usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0+0x49ccc)

Direct leak of 9408 byte(s) in 98 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a8483450f  (/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0+0x1fd50f)

Direct leak of 8403 byte(s) in 266 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a847b787c in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) (/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0+0x18087c)

Direct leak of 7479 byte(s) in 283 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a677c9ee2 in Ogre::GLSupport::initialiseExtensions() (/usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0+0xa6ee2)

Direct leak of 3280 byte(s) in 1 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a849bdbaf  (/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0+0x386baf)

Direct leak of 2020 byte(s) in 59 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a86288afe in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/lib/x86_64-linux-gnu/libstdc++.so.6+0x142afe)

Direct leak of 1344 byte(s) in 8 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a8497dddf  (/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0+0x346ddf)

Direct leak of 1314 byte(s) in 59 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a67769b4c in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) (/usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0+0x46b4c)

Direct leak of 1088 byte(s) in 8 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a6776ed2b in Ogre::GLSL::GLSLLinkProgramManager::getActiveLinkProgram() (/usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0+0x4bd2b)

Direct leak of 797 byte(s) in 1 object(s) allocated from:
    #0 0x7f7a878fcbc8 in malloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10dbc8)
    #1 0x7f7a723d9187  (<unknown module>)

Direct leak of 768 byte(s) in 8 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a8497de2f  (/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0+0x346e2f)

Direct leak of 640 byte(s) in 1 object(s) allocated from:
    #0 0x7f7a878fcffe in __interceptor_realloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10dffe)
    #1 0x7f7a84500490 in XGetVisualInfo (/lib/x86_64-linux-gnu/libX11.so.6+0x3c490)

Direct leak of 593 byte(s) in 26 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a848345cc  (/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0+0x1fd5cc)

Direct leak of 576 byte(s) in 6 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
[... 1372 lines cut ...]
    #6 0x7f7a6367bf5c in (anonymous namespace)::ProxyExec0::ProxyExec0() (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x17c6f5c)
    #7 0x7f7a6367c650 in __static_initialization_and_destruction_0(int, int) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x17c7650)
    #8 0x7f7a6367c6fc in _GLOBAL__sub_I_orbit_view_controller.cpp (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x17c76fc)
    #9 0x7f7a8826eb89  (/lib64/ld-linux-x86-64.so.2+0x11b89)

Indirect leak of 20 byte(s) in 1 object(s) allocated from:
    #0 0x7f7a878fe947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f7a8742a27c in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x51f27c)
[... cut ...] iz_default_plugin.so+0x13ddd1c)
    #5 0x7f7a632857e2 in void class_loader::impl::registerPlugin<rviz::MarkerDisplay, rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x13d07e2)
    #6 0x7f7a6327caf2 in (anonymous namespace)::ProxyExec0::ProxyExec0() (/home/flynn/dev/ros/noetic/devel/lib/librv

flynneva, 2020-06-13:
now with ogre 1.12

<details><summary>rosrun rviz rviz</summary>
<p>

```
[ INFO] [1592019381.969945666]: rviz version 1.14.0
[ INFO] [1592019381.970046262]: compiled against Qt version 5.12.8
[ INFO] [1592019381.970070668]: compiled against OGRE version 1.12.4 (Rhagorthua)
[ INFO] [1592019381.995239325]: Forcing OpenGl version 0.
[ INFO] [1592019382.262089622]: Stereo is NOT SUPPORTED
[ INFO] [1592019382.262395223]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1592019382.285713518]: Error: ScriptCompiler - unexpected token in liberation_sans.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
[ERROR] [1592019422.440086804]: No techniques available for material [PointCloudMaterial0Sphere]
[ERROR] [1592019422.440461544]: No techniques available for material [PointCloudMaterial0FlatSquare]
[ERROR] [1592019422.508575093]: No techniques available for material [PointCloudMaterial1Sphere]
[ERROR] [1592019422.508991574]: No techniques available for material [PointCloudMaterial1FlatSquare]
[ERROR] [1592019422.602285476]: No techniques available for material [PointCloudMaterial2Sphere]
[ERROR] [1592019422.602702484]: No techniques available for material [PointCloudMaterial2FlatSquare]
=================================================================
==2877==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020004921b0 at pc 0x7f695cde9bf7 bp 0x7ffdd5926400 sp 0x7ffdd59263f0
READ of size 4 at 0x6020004921b0 thread T0
    #0 0x7f695cde9bf6 in rviz::SelectionManager::unpackColors(Ogre::PixelBox const&, std::vector<unsigned int, std::allocator<unsigned int> >&) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x5cfbf6)
    #1 0x7f695cdea235 in rviz::SelectionManager::renderAndUnpack(Ogre::Viewport*, unsigned int, int, int, int, int, std::vector<unsigned int, std::allocator<unsigned int> >&) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x5d0235)
    #2 0x7f695cdeda10 in rviz::SelectionManager::pick(Ogre::Viewport*, int, int, int, int, boost::unordered::unordered_map<unsigned int, rviz::Picked, boost::hash<unsigned int>, std::equal_to<unsigned int>, std::allocator<std::pair<unsigned int const, rviz::Picked> > >&, bool) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x5d3a10)
    #3 0x7f693b584420 in rviz::InteractionTool::updateFocus(rviz::ViewportMouseEvent const&) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x17c1420)
    #4 0x7f693b58535f in rviz::InteractionTool::processMouseEvent(rviz::ViewportMouseEvent&) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x17c235f)
    #5 0x7f695ceaced5 in rviz::VisualizationManager::handleMouseEvent(rviz::ViewportMouseEvent const&) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x692ed5)
    #6 0x7f695cd93910 in rviz::RenderPanel::onRenderWindowMouseEvents(QMouseEvent*) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x579910)
    #7 0x7f695cb93720 in rviz::RenderPanel::mouseMoveEvent(QMouseEvent*) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x379720)
    #8 0x7f695c32c2b5 in QWidget::event(QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ad2b5)
    #9 0x7f695c2e9a65 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16aa65)
    #10 0x7f695c2f3342 in QApplication::notify(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x174342)
    #11 0x7f695bebc939 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x286939)
    #12 0x7f695c2f2456 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x173456)
    #13 0x7f695c34835c  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1c935c)
    #14 0x7f695c34b1eb  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1cc1eb)
    #15 0x7f695c2e9a65 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16aa65)
    #16 0x7f695c2f30ef in QApplication::notify(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1740ef)
    #17 0x7f695bebc939 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x286939)
    #18 0x7f6958cb07d2 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x1267d2)
    #19 0x7f6958cb210a in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x12810a)
    #20 0x7f6958c8c35a in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x10235a)
    #21 0x7f694ffb932d  (/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5+0x7932d)
    #22 0x7f6958516fbc in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x51fbc)
    #23 0x7f695851723f  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x5223f)
    #24 0x7f69585172e2 in g_main_context_iteration (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x522e2)
    #25 0x7f695bf14564 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2de564)
    #26 0x7f695bebb4da in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2854da)
    #27 0x7f695bec3245 in QCoreApplication::exec() (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x28d245)
[... cut ...]
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==2877==ABORTING
```
</p>
</details>

symptoms observed:
- crashes immediately if cursor moves within 3d view panel
- only displays pointcloud if 'points' is selected
- cannot resize points if rviz is autosized to fit screen

rhaschke, 2020-06-13:
Regarding Ogre 1.9: Most leakages are external to rviz.
Regarding Ogre 1.12:
* Immediate crash: Did you cherry-picked #1519 on top of noetic-devel? This should exactly fix the crash. I was able to reproduce this particular issue on another machine and it fixed it.
* only display pointcloid if 'points' is selected
  Try running rviz with `--opengl 210` and `--opengl 300`
* cannot resize points if rviz is autosized to fit screen: No idea what happens there... 
Maybe check with `--opengl xxx` as well?

flynneva, 2020-06-13:
that PR is into/from melodic-devel which doesnt include the changes you've made on  noetic-devel that allows for compilation with ogre 1.12.

I get the following error when trying to compile those new changes with ogre 1.12:

```
/home/flynn/dev/ros/noetic/src/rviz/src/rviz/view_controller.h:46:7: error: using typedef-name ‘Ogre::Vector3’ after ‘class’
   46 | class Vector3;
      |       ^~~~~~~
In file included from /usr/include/OGRE/OgreVector.h:32,
                 from /usr/include/OGRE/OgreVector3.h:1,
                 from /home/flynn/dev/ros/noetic/src/rviz/src/rviz/frame_manager.h:39,
                 from /home/flynn/dev/ros/noetic/build/rviz/src/rviz/rviz_autogen/EWIEGA46WW/moc_frame_manager.cpp:9,
                 from /home/flynn/dev/ros/noetic/build/rviz/src/rviz/rviz_autogen/mocs_compilation.cpp:8:
/usr/include/OGRE/OgrePrerequisites.h:253:29: note: ‘Ogre::Vector3’ has a previous declaration here
  253 |     typedef Vector<3, Real> Vector3;
      |                             
```

flynneva, 2020-06-13:
you were right I didnt have those changes included yet.
I tried to just clone the pr but realized it was for melodic and not noetic after a few compilation errors.
manually merged the new commits into noetic-devel.

<details><summary>git log -2 of my current rviz source</summary>
<p>

```
git log -2
commit debf9203e8ecc64f330abfffac985bb6ad8ed08e (HEAD -> noetic-devel)
Merge: c731eff1 6b254a67
Author: Evan Flynn <evanflynn.msu@gmail.com>
Date:   Sat Jun 13 06:59:03 2020 -0700

    Merge commit 'refs/pull/1519/head' of github.com:ros-visualization/rviz into noetic-devel

commit 6b254a67282a05b8cf05b320ee80a58f224b2364 (fix-memory-access)
Author: Robert Haschke <rhaschke@techfak.uni-bielefeld.de>
Date:   Sat Jun 13 07:40:58 2020 +0200

    change private API instead of const-casting
```

<p>
</details>

after compilation....

<details><summary>rosrun rviz rviz</summary>
<p>

```
rosrun rviz rviz
[ INFO] [1592057674.377568903]: rviz version 1.14.0
[ INFO] [1592057674.377715849]: compiled against Qt version 5.12.8
[ INFO] [1592057674.377768214]: compiled against OGRE version 1.12.4 (Rhagorthua)
[ INFO] [1592057674.407054089]: Forcing OpenGl version 0.
[ INFO] [1592057674.661664123]: Stereo is NOT SUPPORTED
[ INFO] [1592057674.661984415]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1592057674.687852394]: Error: ScriptCompiler - unexpected token in liberation_sans.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
[ERROR] [1592057740.081238943]: No techniques available for material [PointCloudMaterial0Sphere]
[ERROR] [1592057740.081565951]: No techniques available for material [PointCloudMaterial0FlatSquare]
[ERROR] [1592057740.138587982]: No techniques available for material [PointCloudMaterial1Sphere]
[ERROR] [1592057740.138724765]: No techniques available for material [PointCloudMaterial1FlatSquare]
AddressSanitizer:DEADLYSIGNAL
=================================================================
==27135==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000040 (pc 0x7f97b539aa04 bp 0x7ffd4778b3d0 sp 0x7ffd4778b2b8 T0)
==27135==The signal is caused by a READ memory access.
==27135==Hint: address points to the zero page.
    #0 0x7f97b539aa03 in Ogre::UserObjectBindings::getUserAny(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x3ada03)
    #1 0x7f97b7e4ca5b in rviz::SelectionManager::handleSchemeNotFound(unsigned short, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::Material*, unsigned short, Ogre::Renderable const*) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x5d5a5b)
    #2 0x7f97b521c157 in Ogre::MaterialManager::_arbitrateMissingTechniqueForActiveScheme(Ogre::Material*, unsigned short, Ogre::Renderable const*) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x22f157)
    #3 0x7f97b5217ec5 in Ogre::Material::getBestTechnique(unsigned short, Ogre::Renderable const*) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x22aec5)
    #4 0x7f97b529bc16 in Ogre::RenderQueue::addRenderable(Ogre::Renderable*, unsigned char, unsigned short) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2aec16)
    #5 0x7f97b7d70d6e in rviz::PointCloud::_updateRenderQueue(Ogre::RenderQueue*) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x4f9d6e)
    #6 0x7f97b529b6bb in Ogre::RenderQueue::processVisibleObject(Ogre::MovableObject*, Ogre::Camera*, bool, Ogre::VisibleObjectsBoundsInfo*) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2ae6bb)
    #7 0x7f97b52fa9a8 in Ogre::SceneNode::_findVisibleObjects(Ogre::Camera*, Ogre::RenderQueue*, Ogre::VisibleObjectsBoundsInfo*, bool, bool, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x30d9a8)
    #8 0x7f97b52fa9fe in Ogre::SceneNode::_findVisibleObjects(Ogre::Camera*, Ogre::RenderQueue*, Ogre::VisibleObjectsBoundsInfo*, bool, bool, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x30d9fe)
    #9 0x7f97b52fa9fe in Ogre::SceneNode::_findVisibleObjects(Ogre::Camera*, Ogre::RenderQueue*, Ogre::VisibleObjectsBoundsInfo*, bool, bool, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x30d9fe)
    #10 0x7f97b52e0cf6 in Ogre::SceneManager::_findVisibleObjects(Ogre::Camera*, Ogre::VisibleObjectsBoundsInfo*, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2f3cf6)
    #11 0x7f97b52e9685 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2fc685)
    #12 0x7f97b515d9f9 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x1709f9)
    #13 0x7f97b52b70e4 in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2ca0e4)
    #14 0x7f97b52b6e91 in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2c9e91)
    #15 0x7f97b52b6cc2 in Ogre::RenderTarget::updateImpl() (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2c9cc2)
    #16 0x7f97b52b7487 in Ogre::RenderTarget::update(bool) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2ca487)
    #17 0x7f97b7e48c05 in rviz::SelectionManager::render(Ogre::Viewport*, Ogre::SharedPtr<Ogre::Texture> const&, int, int, int, int, Ogre::PixelBox&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int) (/home/flynn/dev/ros/noetic/devel/lib/librviz.so+0x5d1c05)
[... cut ...]
    #47 0x5598361747cd in _start (/home/flynn/dev/ros/noetic/devel/lib/rviz/rviz+0xc7cd)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x3ada03) in Ogre::UserObjectBindings::getUserAny(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const
==27135==ABORTING
```
</p>
</details>

symptoms observed:
- can now move mouse through 3d window without crash
- when a pointcloud topic is added to visualize, rviz crashes immediately

rhaschke, 2020-06-13:
We are slowly progressing. :smile: Did you try to run rviz with `--opengl 300` or `--opengl 210` as well??

flynneva, 2020-06-13:
when running with either `--opengl 300` or `--opengl 210` :
- when pointcloud topic is added to visualize, rviz *does not* crash!
- can change size of any points type when rviz is not maximized to fit window
- when maximized to fit window, only 'points' type is not able to change point size.

<details><summary>rosrun rviz rviz --opengl 300</summary>
<p>

```
[ INFO] [1592058924.569122316]: rviz version 1.14.0
[ INFO] [1592058924.569231356]: compiled against Qt version 5.12.8
[ INFO] [1592058924.569276001]: compiled against OGRE version 1.12.4 (Rhagorthua)
[ INFO] [1592058924.599036252]: Forcing OpenGl version 3.
[ INFO] [1592058924.828757378]: Stereo is NOT SUPPORTED
[ INFO] [1592058924.828992698]: OpenGl version: 3 (GLSL 1.3).
[ERROR] [1592058924.850943315]: Error: ScriptCompiler - unexpected token in liberation_sans.fontdef(1): 'Liberation'. If this is a legacy script you must prepend the type (e.g. font, overlay).
^C
=================================================================
==27584==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4416 byte(s) in 92 object(s) allocated from:
    #0 0x7f32c34a1947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f32a56044e6 in Ogre::FreeImageCodec::startup() (/usr/lib/x86_64-linux-gnu/OGRE-1.12/Codec_FreeImage.so.1.12.4+0x64e6)
    #2 0x7f32c0525aad in Ogre::Root::loadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x2eaaad)
    #3 0x6e6967756c4f  (<unknown module>)

Direct leak of 1200 byte(s) in 30 object(s) allocated from:
    #0 0x7f32c34a1947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f32c03ce6b9 in Ogre::ConvexBody::_initialisePool() (/lib/x86_64-linux-gnu/libOgreMain.so.1.12.4+0x1936b9)

Direct leak of 797 byte(s) in 1 object(s) allocated from:
    #0 0x7f32c349fbc8 in malloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10dbc8)
    #1 0x7f32af8db187  (<unknown module>)

Direct leak of 640 byte(s) in 1 object(s) allocated from:
    #0 0x7f32c349fffe in __interceptor_realloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10dffe)
    #1 0x7f32c0104490 in XGetVisualInfo (/lib/x86_64-linux-gnu/libX11.so.6+0x3c490)

Direct leak of 256 byte(s) in 1 object(s) allocated from:
    #0 0x7f32c349fbc8 in malloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10dbc8)
    #1 0x7f32b7ada2f4  (/lib/x86_64-linux-gnu/libfontconfig.so.1+0x212f4)

Direct leak of 160 byte(s) in 1 object(s) allocated from:
    #0 0x7f32c34a1947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f32a1135b86 in void class_loader::impl::registerPlugin<rviz::DepthCloudDisplay, rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x10c7b86)
    #2 0x7f32a111d2d0 in (anonymous namespace)::ProxyExec0::ProxyExec0() (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x10af2d0)
    #3 0x7f32a111f569 in __static_initialization_and_destruction_0(int, int) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x10b1569)
    #4 0x7f32a111ff72 in _GLOBAL__sub_I_depth_cloud_display.cpp (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x10b1f72)
    #5 0x7f32c3e11b89  (/lib64/ld-linux-x86-64.so.2+0x11b89)

Direct leak of 160 byte(s) in 1 object(s) allocated from:
    #0 0x7f32c34a1947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f32a129cb9c in void class_loader::impl::registerPlugin<rviz::FluidPressureDisplay, rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x122eb9c)
    #2 0x7f32a129af60 in (anonymous namespace)::ProxyExec0::ProxyExec0() (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x122cf60)
    #3 0x7f32a129b5d1 in __static_initialization_and_destruction_0(int, int) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x122d5d1)
    #4 0x7f32a129b7e0 in _GLOBAL__sub_I_fluid_pressure_display.cpp (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x122d7e0)
    #5 0x7f32c3e11b89  (/lib64/ld-linux-x86-64.so.2+0x11b89)

Direct leak of 160 byte(s) in 1 object(s) allocated from:
    #0 0x7f32c34a1947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
    #1 0x7f32a13bc5ce in void class_loader::impl::registerPlugin<rviz::LaserScanDisplay, rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x134e5ce)
    #2 0x7f32a13ba400 in (anonymous namespace)::ProxyExec0::ProxyExec0() (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x134c400)
    #3 0x7f32a13baa71 in __static_initialization_and_destruction_0(int, int) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x134ca71)
    #4 0x7f32a13bac80 in _GLOBAL__sub_I_laser_scan_display.cpp (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x134cc80)
[... cut ...]
    #3 0x7f32a1708a4a in __static_initialization_and_destruction_0(int, int) (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x169aa4a)
    #4 0x7f32a1708c1a in _GLOBAL__sub_I_pose_with_covariance_display.cpp (/home/flynn/dev/ros/noetic/devel/lib/librviz_default_plugin.so+0x169ac1a)
    #5 0x7f32c3e11b89  (/lib64/ld-linux-x86-64.so.2+0x11b89)

Direct leak of 64 byte(s) in 2 object(s) allocated from:
    #0 0x7f32c349fbc8 in malloc (/lib/x86_64-linux-gnu/libasan.so.5+0x10dbc8)
    #1 0x7f32b442f3cf in XextAddDisplay (/lib/x86_64-linux-gnu/libXext.so.6+0xe3cf)
```
</p>
</details>

rhaschke, 2020-06-13:
Thanks. I will have a look later on the 3rd issue. I cannot explain why rviz behaves differently in maximized view. Is this only affecting to auto-maximized view / snapping in the gnome desktop or also when you explicitly maximize via the menu?
Maybe as a workaround, we should switch to opengl 3.0 on mesa by default if that solves most issues. Will need to check where this is handled...

flynneva, 2020-06-13:
yeah it happens both when snapping & when I explicitly maximize via the menu. its strange.

simonschmeisser, 2020-06-13:
window manager sometimes handle full screen windows differently (turning off compositing), but we're talking about maximized and not full screen right?

flynneva, 2020-06-13:
this might help explain what i'm seeing....

[image]

flynneva, 2020-06-13:
full screen example....

[image]

rhaschke, 2020-06-24:
I have no idea regarding the maximize-issue. I cannot reproduce that problem on my machines.
Regarding the OpenGL issue, I noticed that the OpenGL version is only used to switch between two shader source variants, namely 1.2 and 1.5:
https://github.com/ros-visualization/rviz/blob/291df89957e2f643a3fde28e5cb239fe1f6f5b73/src/rviz/ogre_helpers/render_system.cpp#L289-L300

So, I guess the 1.5 shader sources are buggy or incompatible with Mesa OpenGL.
@flynneva, I'm expecting that rviz fails on Ubuntu Bionic / ROS Melodic as well. Can you confirm?
If not, i.e. everything works as expected there, I'm curious about the Mesa OpenGL version used in Melodic.

flynneva, 2020-06-26:
before I upgraded to Ubuntu 20.04 I used ROS Melodic on Ubuntu 18.04. it worked perfect on my machine and I never had any issues.

only after upgrading I observed this issue.

rhaschke, 2020-06-28:
I installed Ubuntu 20.04 on my laptop today as well (before I used an older machine only). rviz reports OpenGL version 4.6 as well now, but I don't see any issues regarding rendering of pointcloud data. I'm clueless for the remaining issues.

theseankelly, 2020-07-02:
I too am experiencing this. I have had the same laptop (Lenovo T470p) using integrated Intel graphics. Haven't had any issues with `PointCloud2` on 16.04/Kinetic, 18.04/Melodic. On 20.04/Noetic I experience the giant pixel size and tremendous CPU load when viewing point cloud information. Let me know if there's info worth providing.

simonschmeisser, 2020-07-02:
What desktop environment/compositor are you using? @rhaschke seems to be using Plasma/KDE/kwin (with a really old-school theme ;) ) while all other screenshots show the ubuntu one (gnome 3 ?)

Also, as said before, reporters seems to have newer generations of intel HW so maybe some fallback is not automatically enabled there and we should request it explicitly (in the shaders I guess)

theseankelly, 2020-07-02:
I'm using vanilla Ubuntu 20.04 with gnome3. My T470p has an Intel i7-7700HQ. It's also got a discrete NVIDIA card, but I have it disabled for power consumption reasons. Not sure if that factors in here.

I'm happy to try hacking on some stuff including building RVIZ from source withat various opitons...but I don't know much about the Linux graphics stack and would need some guidance. Let me know!

rhaschke, 2020-07-02:
I am using the xfce desktop (of xubuntu) - very slim :wink:
But I also tried gnome. No issues in both cases. I guess, @simonschmeisser, you don't experience problems as well? Are you using a native Ubuntu 20.04 system at all? My impression is that the Mesa lib shipped with Ubuntu 20.04 is causing the issues.
But I have no idea why Mesa rendering works on some machines and doesn't on others. All users have reported OpenGL 4.6 so far.
Maybe affected users could try to downgrade their Mesa OpenGL lib?

AndreasR30, 2020-09-02:
I experience the same issues. My graphics output is also Intel-only since I use my GPU just for computations. I have the same issues as above. Using "Points" as style I get huge balls like in the image. Additionally, when I use "Flat Squares" I get no color. All of them are white/grey.
Both issues go away with the --opengl 210 Flag.

RobustControl, 2020-11-15:
"> rviz --opengl 210"
This command is quite useful. It can also solve my problem.
I use ubuntu 20.04 and ROS noetic.

> [ INFO] [1605453384.934870437]: rviz version 1.14.1
[ INFO] [1605453384.934922755]: compiled against Qt version 5.12.8
[ INFO] [1605453384.934935759]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1605453384.944656582]: Forcing OpenGl version 0.
[ INFO] [1605453385.199014948]: Stereo is NOT SUPPORTED
[ INFO] [1605453385.199129051]: OpenGl version: 4.6 (GLSL 4.6).

I use rviz to display PointCloud2 with Style=Boxes, it can only display some gray points. After I use this command, the display is normal.
Also, when I use visualization_msgs::Marker with type=CUBE_LIST, rviz can only display a point for a cube. After I use this command, the display is normal too.

swiz23, 2021-02-16:
Just ran into this same issue. I have a fully updated Ubuntu 20.04 / ROS Noetic system, and the pointclouds are gigantic! This is only with the 'Points' selection, but with the flat squares or squares, etc... , it looks just fine.

rviz version 1.14.4
compiled against Qt version 5.12.8
compiled against OGRE version 1.9.0 (Ghadamon)
Forcing OpenGl version 0.
Stereo is NOT SUPPORTED
OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.

The computer is a NUC7i5BNH

rhaschke, 2021-02-16:
@swiz23, I'm surprised that you still face this issue, although the broken Mesa system was detected and GLSL was downgraded to 1.4. Could you please try to enforce a later OpenGL version manually:
`rviz --opengl 210`

flynneva, 2021-02-16:
@rhaschke for the record I still see this issue too.

[24 more replies]

PR #1559 (2020-11-16): Enforce GLSL 1.4 on Mesa systems
Fixes #1508. There seems to be an issue with GLSL 1.5 material scripts on Mesa-based systems.
closed 2021-08-22 by rhaschke