ros-visualization/rviz #1509: [noetic] Broken 2D Nav Goal button (orientation)

doisyg, 2020-05-27:
Last deb version:
```
[ INFO] [${node}]: rviz version 1.14.0
[ INFO] [${node}]: compiled against Qt version 5.12.8
[ INFO] [${node}]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [/rviz_1590590368092672263]: Forcing OpenGl version 0.
[ INFO] [/rviz_1590590368092672263]: Stereo is NOT SUPPORTED
[ INFO] [/rviz_1590590368092672263]: OpenGl version: 4.6 (GLSL 4.6).
```
The 2D Nav Goal button seems broken under noetic. No orientation is published on the topic:
```
rostopic echo /move_base_simple/goal
header: 
  seq: 0
  stamp: 
    secs: 1590590378
    nsecs:  90359437
  frame_id: "map"
pose: 
  position: 
    x: 0.4193878173828125
    y: 0.5008087158203125
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0
---

```

Though the angle is displayed in the console:
`Setting goal: Frame:map, Position(0.419, 0.501, 0.000), Orientation(0.000, 0.000, 0.000, 0.000) = Angle: -2.977`

doisyg, 2020-05-27:
Same issue for the 2D Pose Estimate button:
```
rostopic echo /initialpose
header: 
  seq: 0
  stamp: 
    secs: 1590596724
    nsecs: 931619017
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: -0.12504196166992188
      y: 0.07902240753173828
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 0.0
  covariance: [0.25, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.25, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.06853892326654787]
---
```
with:
`[ INFO] [/rviz_1590596700755421369]: Setting pose: -0.125 0.079 -2.567 [frame=map]`

doisyg, 2020-05-27:
Just submitted a PR to fix it https://github.com/ros-visualization/rviz/pull/1510. The wrong TF2 conversion side was used for creating the quaternion part from theta and the tf2::Quaternion

rhaschke, 2020-05-29:
Fixed via #1510.

closed 2020-05-29 by rhaschke

CodeFinder2, 2020-06-08:
@rhaschke Thanks for fix, affects me as well! Any plans when this will be available in the Ubuntu packages? Kinda critical for me ... :see_no_evil: